PX4/PX4-Autopilot #11036: Incremental build issues for IO

bkueng, 2018-12-14:
- incremental build for the IO is broken, i.e. a change to the IO firmware code is not included in a subsequent build.
- `make/ninja clean` in `build/px4_fmu-v5_default` does not clean the IO build. However a subsequent build rebuilds the IO.

dagar, 2018-12-14:
The way this works is the px4io driver initiates the px4_io-v2 build. Previously this was done by calling make directly from a cmake custom command. Recently I changed it to use cmake ExternalProject (https://cmake.org/cmake/help/latest/module/ExternalProject.html) to work around a number of cmake IDE related issues.

The dependencies were never quite right in the previous setup. I'll look for a way we can use the real underlying dependencies with ExternalProject so that final px4 binary isn't rebuilt constantly.

bresch, 2019-03-28:
The issue is still there: every time I want to test a modification I did in px4iofirmware, I have to do a `make clean` before I rebuild to include the changes.

PR #12805 (2019-08-25): px4io: force io firmware build
This fixes incremental build of the px4io firmware on targets that use the px4io driver.

fixes #11036 #11042
closed 2019-08-25 by dagar