PX4/PX4-Autopilot #11045: PREFLIGHT FAIL: NO MAG SENSOR 0

Addy0265, 2018-12-14:
Hi all,
I just got a new Intel Aero RTF drone. Upon updating the flight controller firmware to 1.6.5, I get a preflight fail error in QGroundControl while calibrating the controller. I've tried updating the firware to the stable releases 1.7.3 and 1.8.1 but still I get the same error.
The flight controller used to work out of the box without any updates, so I am pretty sure the hardware is fine.
I found out by looking into other threads on this forum that this might happen because of the firmware version and 1.6.5 was concluded to be a stable version which had everything working fine. But, it doesnt work for me.

Kindly help!

AlexisTM, 2018-12-16:
Can you ensure your external GPS/mag is connected?

Addy0265, 2018-12-17:
Yes, I haven't touched the hardware.
It was working before updating the firmware (All sensors were fine, I even operated it using a RC)  and just after updating it, while running calibration in QGroundControl, it throws this error.

julianoes, 2019-07-09:
@Addy0265 is this still a problem with 1.9?

Addy0265, 2019-07-15:
> 
> 
> @Addy0265 is this still a problem with 1.9?

Hi, the issue was fixed.
There was sme connection issue with the GPS/MAG module.
I tried re-soldering the connections but it still didn't work.
So I replaced it wit another GPS module of the same make and I was able to calibrate it.

closed 2019-07-15 by Addy0265