PX4/PX4-Autopilot #11047: Direct fetching of that commit failed.

dayL-W, 2018-12-15:
Hello everyone.when I build the project,it happend en error 
error: Server does not allow request for unadvertised object efc3d9e92bd10623ab3025f012fd7d481e1a7267
Fetched in submodule path 'platforms/nuttx/NuttX/nuttx', but it did not contain efc3d9e92bd10623ab3025f012fd7d481e1a7267. Direct fetching of that commit failed.

this is what I do:
1 clone the Firmware
2 git checkout v1.7.4beta
3 clone the submodule: git submodule update --init --recursive

could you tell me what should I do?
thank you

dagar, 2018-12-15:
I wouldn't use an old beta version. Try the current stable v1.8.2 or master (soon to be v1.9.0).

dayL-W, 2018-12-15:
thank you very much !!! when I run
` make submodulesclean
git submodule update --init --recursive
`
It works!!