PX4/PX4-Autopilot #11061: Inconsistent heading estimation error / large offset

DavidMackie, 2018-12-17:
**Description**
Using the latest stable release of PX4 and latest stable release of QGroundControl, there are unexpected heading estimations from the compass. Usually a heading error of 90-180 degrees is introduced, then the heading starts drifting around.

**System and steps to reproduce**
1. Drotek - Dropix board
2. Drotek - Ublox NEO-M8N GPS + LIS3MDL compass
3. PX4 latest stable firmware - DeltaQuad or Generix quad x airframes
4. Calibration done using QGC 
5. Board orientation set to ROLL 90 - YAW 90
6. Compass orientation set to None, compass aligned with forward flight direction

Numerous tests were done and the heading offset appears inconsistently. The tests were done in varied environments (inside/in a field), different firmware (stable, DQRC12E, master), different boards of the same type.

**Tests and Log Files**

Flight tests were done last week to reproduce the problem which occurred during 2 out of 6 flights. Flights are in multicopter mode. No changes to the system were done between the flights. The heading problem was not present on the ground.

The raw mag data was logged during the flights.

Here is the link to the 180 degree offset. The drifting begins at approximately 9 minutes into the recording. the heading of the drone was approximately 100 degrees throughout the majority of the flight time.
[180 degree offset flight - PX4 Flight Review](https://review.px4.io/plot_app?log=d3dda4b9-7fba-4b71-bcb1-168a71d6fd61).

Here is the link to one of our flights with no heading issues;
[No issue flight - PX4 Flight Review](https://review.px4.io/plot_app?log=3d3bb913-1826-4488-9bf0-42fddaf1c66e).

Thanks for any help.

PS: The true problem is the inconsistency of when it occurs. Hard to rule out software or specific components, because it 'sometimes' happens for different boards,, different compasses, different PX4 version and even the environment.
It would be really helpful if we could rule out specific parts of PX4.

julianoes, 2019-07-09:
Sorry for the late response here, we need to look into this.

julianoes, 2019-10-07:
Thanks for the issue.

Is this the same as https://github.com/PX4/Firmware/issues/10655 and https://github.com/PX4/Firmware/issues/10951?