PX4/PX4-Autopilot #11065: logger orb subscribe multi error

cimbar, 2018-12-18:
**Description**
In flight logs, orb_subscribe_multi error messages come out with many arguments. It relates with logger and there is no error message on QGC while flying. There wasn't this error at older flights with same SD card and the SD card is Samsung EVO so it shouldn't be a SD card read/write issue. Both v1.8.0 and v.1.8.2 Firmware versions were tried and same result. I couldn't see any influence of this error on flight but it is worrying. Is there any idea what's the source of this error and how can i solve this issue?

**Log Files and Screenshots**
Such examples:
https://logs.px4.io/plot_app?log=6cf7f9bf-fc44-4ea5-a2fd-1caff84fd214
https://logs.px4.io/plot_app?log=6f76d5cd-4698-4870-bedb-469bc7a5f1f5
[image]

Louisdirk, 2018-12-29:
@cimbar. I had the same issue. @davids5, could you help to figure out the reason?

davids5, 2018-12-29:
@cimbar , @Louisdirk - it look like there are not enough FD or there is a memory corruption that makes it seem that way. 

@dagar are there a parameters to turn on and off different logging streams? 

@cimbar posting your parameters would help troubleshoot this.

You can try to increase [CONFIG_NFILE_DESCRIPTORS](https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/nuttx-config/nsh/defconfig#L784)  by increments of 4 and see if that solves your problem. But keep in mind that may result in lower free RAM.

dagar, 2018-12-29:
Yes it's `SDLOG_PROFILE` that controls the groups. You can't log everything at the same time unfortunately.

closed 2019-01-07 by bkueng