PX4/PX4-Autopilot #11070: Mavlink rate control still doesn't follow target rate

jasonbeach, 2018-12-18:
**Describe the bug**
I would like to stream HIGHRES_IMU and ATTITUDE at 200Hz, but adding these lines in etc/extras.txt results in them streaming at ~160 - 170 Hz:

mavlink start -d /dev/ttyACM0 -b 921600 -m onboard -r 80000
mavlink stream -d /dev/ttyACM0 -s ATTITUDE -r 200
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200

It seems exactly like 
https://github.com/PX4/Firmware/issues/3291
https://github.com/PX4/Firmware/issues/6235
which are closed.

**To Reproduce**
Running v1.8.1 with Nuttx 7.22.0 on a pixracer, but issue seems to also occur on a pixhawk.  On the sdcard I create the etc/extras.txt as described above and then monitor the message publish rate via either QGroundControl or mavros. I'm connected to the autopilot via usb cable, but using an FTDI cable on telem2 doesn't seem to change thing.

If I increase the stream rate from 200 to 250 (and make no other changes) then the rate increases to about ~230Hz

**Expected behavior**
I expect the messages to be published at the requested rate

**Log Files and Screenshots**
mavlink console screenshots
https://www.dropbox.com/s/6ppp4ro1qwli0y1/slowMavlinkStatus.png?dl=0
https://www.dropbox.com/s/i82su7saow26pxa/slowMavlinkStatus2.png?dl=0
https://www.dropbox.com/s/yv9zb4houa1u3ws/mavlink_inspector.png?dl=0

siddhu45, 2019-03-12:
If you say, you're using telem2 connection, what baudrate is it set at? try setting it at 921600 and then launch mavros on this connection. Also, change the `/dev/ttyACM0` to `/dev/ttyS2` on your etc/extras.txt file as telem2 is a serial communication.

jasonbeach, 2019-07-02:
I've changed the `/dev/ttyACM0` to `/dev/ttyS2` and have the baudrate set at 921600 and the issue persists.

julianoes, 2019-07-02:
When this happens, can you paste the output of `mavlink status`?

jasonbeach, 2019-07-10:
My apologies for not responding sooner.  Here's the output of `ver all` 

```
HW arch: PX4_FMU_V2
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 5c8b5f10b5b1b11e9b0ed2d01423382cace49978
FW version: 1.9.0 80 (17367168)
FW git-branch: master-v2-frsky
OS: NuttX
OS version: Release 7.28.0 (119275775)
OS git-hash: ea24b543df6aab6607e1fa758e4a7586b9ed3113
Build datetime: Apr 12 2019 12:13:48
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0001000000003630303033334707002e002b
MCU: STM32F42x, rev. Y

WARNING   WARNING   WARNING!
Revision Y has a silicon errata:
This device can only utilize a maximum of 1MB flash safely!
https://docs.px4.io/en/flight_controller/silicon_errata.html
```
and the contents of `etc/extras.txt`
```
mavlink start -d /dev/ttyS2 -b 921600 -m onboard -r 80000
mavlink stream -d /dev/ttyS2 -s HIGHRES_IMU -r 245
mavlink stream -d /dev/ttyS2 -s ATTITUDE_QUATERNION -r 245
camera_feedback start
```
and finally what you asked for, the result of `mavlink status`
```
instance #0:
	mavlink chan: #0
	type:		3DR RADIO
	  rssi:		32
	  remote rssi:	0
	  txbuf:	95
	  noise:	41
	  remote noise:	0
	  rx errors:	0
	  fixed:	0
	flow control: ON
	rates:
	  tx: 0.404 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 0.403
	  tx rate max: 1200 B/s
	  rx: 0.034 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 1
	transport protocol: serial (/dev/ttyS1 @57600)

instance #1:
	GCS heartbeat:	22389 us ago
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 41.916 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 80000 B/s
	  rx: 0.047 kB/s
	FTP enabled: NO, TX enabled: YES
	mode: Onboard
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS2 @921600)
	ping statistics:
	  last: 11.68 ms
	  mean: 10.41 ms
	  max: 27.55 ms
	  min: 2.44 ms
	  dropped packets: 0
```
and the rates I see:
[image]
This is on a vehicle that has an orginal pixhawk autopilot, but I've seen the same behavior with a pixracer.

julianoes, 2019-07-14:
I was expecting the rate multiplier to be reduced because you are requesting "too much" and the link would be saturated if it was not throttled. However, the rate multiplier on ttyS2 seems 1, so fine:

```
	  tx rate mult: 1.000
```

This means that either the messages are not sent at 250 Hz, or that some are dropped on the way, or that QGC misses some. I suspect the first reason but we need to check.

julianoes, 2019-07-14:
I can reproduce this in master SITL. The stream rate doesn't get above about 203 Hz with:
```
mavlink stream -u 14570 -s HIGHRES_IMU -r 500
```
I would expect around 250 Hz because `uorb top` shows `sensor_combined` published at 250 Hz.

julianoes, 2019-07-14:
Ok for SITL this has to do with lockstep somehow. Without lockstep 249 Hz can be reached.

jasonbeach, 2019-07-14:
What is lockstep?

julianoes, 2019-07-14:
> What is lockstep?

https://dev.px4.io/master/en/simulation/#lockstep-simulation

mhkabir, 2019-07-14:
On actual hardware, if you set the stream to `unlimited_rate` in mavlink.cpp, it seems to stream at the correct (topic) rate.

vivekalgorythma, 2019-07-30:
I too am facing this issue. My mavlink extras file is as shown below 

```
mavlink stop-all

mavlink start -d /dev/ttyS1 -b 921600 -r 80000 
mavlink start -d /dev/ttyACM0 -b 921600 -r 80000

mavlink stream -d /dev/ttyS1 -s LOCAL_POSITION_NED -r 100
mavlink stream -d /dev/ttyS1 -s HIGHRES_IMU -r 200
mavlink stream -d /dev/ttyS1 -s ATTITUDE -r 50
mavlink stream -d /dev/ttyS1 -s ALTITUDE -r 50
mavlink stream -d /dev/ttyS1 -s GPS_RAW_INT -r 10
mavlink stream -d /dev/ttyS1 -s UTM_GLOBAL_POSITION -r 10
mavlink stream -d /dev/ttyS1 -s ODOMETRY -r 50

mavlink stream -d /dev/ttyACM0 -s LOCAL_POSITION_NED -r 100
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200
mavlink stream -d /dev/ttyACM0 -s ATTITUDE -r 50
mavlink stream -d /dev/ttyACM0 -s ALTITUDE -r 50
mavlink stream -d /dev/ttyACM0 -s GPS_RAW_INT -r 10
mavlink stream -d /dev/ttyACM0 -s UTM_GLOBAL_POSITION -r 10
mavlink stream -d /dev/ttyACM0 -s ODOMETRY -r 50
```
The autopilot i use is the pixhawk 4 mini. I have tried different versions of the px4 firmware. 

Firmware 1.8.2 with highres_imu @ 200 Hz i get an output rate between 181 - 182 Hz 
Firmware 1.8.2 with highres_imu @ 250 Hz i get an output rate between 221 - 230 Hz
Firmware 1.9.2 with highres_imu @ 200 Hz i get an output rate between 178 - 180 Hz 
Firmware 1.9.2 with highres_imu @ 250 Hz i get an output rate between 225 - 232 Hz

The fact that i get a higher rate when i increase the rate to 250 shows that it is not a data rate saturation issue. 
 Any suggestions from the community on what i can do to get a stable 200 Hz rate?

berndpfrommer, 2021-05-01:
Hitting the same with the pixracer and the new pixracer pro: setting rate to 250 Hz, but only getting 185-195 via mavlink. Has this ever been resolved? What additional data is helpful to find the cause of this issue? Thanks!

dagar, 2021-05-01:
> Hitting the same with the pixracer and the new pixracer pro: setting rate to 250 Hz, but only getting 185-195 via mavlink. Has this ever been resolved? What additional data is helpful to find the cause of this issue? Thanks!

Which rate are you setting? In current master the default IMU rate is 200 Hz, but the actual rate depends on the sensor (rounded multiple). Can you check `sensors status` and `uorb top`?

berndpfrommer, 2021-05-08:
Looking at ``uorb top`` told me that gyro and accel are running at 400Hz, which is consistent with the 2.5ms interval that ``sensors status`` gives. However, the ``sensor_combined`` topic was running at 200Hz because the ``IMU_INTEG_RATE`` rate was set to 200. 

After setting that to 400 I was able to reach up to 333Hz, but not the full 400Hz.

Here is my ``extras.txt`` file.
```
mavlink stream -d /dev/ttyS2 -b 921600 -m onboard -r 100000
mavlink stream -d /dev/ttyS2 -s HIGHRES_IMU -r 400
```

This is the output from ``sensors status``
```
INFO  [sensors] selected gyro: 3670026 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator] 	val:  -0.0021, lp:  -0.0016 mean dev:   0.0000 RMS:   0.0017 conf:   1.0000
INFO  [data_validator] 	val:  -0.0012, lp:  -0.0025 mean dev:  -0.0000 RMS:   0.0009 conf:   1.0000
INFO  [data_validator] 	val:  -0.0056, lp:  -0.0003 mean dev:  -0.0000 RMS:   0.0011 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 50, state: OFF
INFO  [data_validator] 	val:  -0.0019, lp:   0.0023 mean dev:   0.0000 RMS:   0.0031 conf:   1.0000
INFO  [data_validator] 	val:   0.0007, lp:   0.0028 mean dev:  -0.0000 RMS:   0.0030 conf:   1.0000
INFO  [data_validator] 	val:   0.0074, lp:   0.0023 mean dev:   0.0000 RMS:   0.0029 conf:   1.0000
INFO  [data_validator] sensor #2, prio: 50, state: OFF
INFO  [data_validator] 	val:  -0.0033, lp:  -0.0010 mean dev:  -0.0000 RMS:   0.0023 conf:   1.0000
INFO  [data_validator] 	val:   0.0006, lp:   0.0026 mean dev:  -0.0000 RMS:   0.0016 conf:   1.0000
INFO  [data_validator] 	val:   0.0037, lp:   0.0011 mean dev:  -0.0000 RMS:   0.0019 conf:   1.0000

INFO  [sensors] selected accel: 3670026 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 50, state: OK
INFO  [data_validator] 	val:  -0.2928, lp:  -0.2846 mean dev:  -0.0001 RMS:   0.0088 conf:   1.0000
INFO  [data_validator] 	val:  -0.9128, lp:  -0.9049 mean dev:   0.0002 RMS:   0.0111 conf:   1.0000
INFO  [data_validator] 	val:  -9.8995, lp:  -9.8876 mean dev:   0.0011 RMS:   0.0098 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 50, state: OFF
INFO  [data_validator] 	val:  -0.2543, lp:  -0.2620 mean dev:  -0.0001 RMS:   0.0136 conf:   1.0000
INFO  [data_validator] 	val:  -0.7607, lp:  -0.7572 mean dev:   0.0001 RMS:   0.0145 conf:   1.0000
INFO  [data_validator] 	val:  -9.8485, lp:  -9.8533 mean dev:  -0.0001 RMS:   0.0169 conf:   1.0000
INFO  [data_validator] sensor #2, prio: 50, state: OFF
INFO  [data_validator] 	val:  -0.2006, lp:  -0.1927 mean dev:  -0.0002 RMS:   0.0157 conf:   1.0000
INFO  [data_validator] 	val:  -0.7136, lp:  -0.6894 mean dev:  -0.0001 RMS:   0.0161 conf:   1.0000
INFO  [data_validator] 	val:  -9.9281, lp:  -9.9045 mean dev:   0.0008 RMS:   0.0161 conf:   1.0000

INFO  [vehicle_magnetometer] selected magnetometer: 73225 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 75, state: OK
INFO  [data_validator] 	val:  -0.1171, lp:  -0.1164 mean dev:  -0.0001 RMS:   0.0024 conf:   1.0000
INFO  [data_validator] 	val:   0.1968, lp:   0.1975 mean dev:  -0.0009 RMS:   0.0058 conf:   1.0000
INFO  [data_validator] 	val:   0.3945, lp:   0.3920 mean dev:  -0.0017 RMS:   0.0106 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 50, state: OK
INFO  [data_validator] 	val:  -0.2358, lp:  -0.2344 mean dev:  -0.0002 RMS:   0.0089 conf:   1.0000
INFO  [data_validator] 	val:   0.1147, lp:   0.1164 mean dev:   0.0002 RMS:   0.0092 conf:   1.0000
INFO  [data_validator] 	val:   0.5063, lp:   0.5052 mean dev:  -0.0001 RMS:   0.0094 conf:   1.0000
INFO  [sensor_calibration] MAG 73225 EN: 1, offset: [ 0.155  0.039  0.066], scale: [ 1.000  1.000  1.000], External ROT: 0
INFO  [sensor_calibration] MAG 589834 EN: 1, offset: [-0.121 -0.517 -0.444], scale: [ 1.013  1.001  1.003], Internal

INFO  [vehicle_air_data] selected barometer: 6815762 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 0, state: OK
INFO  [data_validator] 	val: 101096.1250, lp: 101096.1094 mean dev: -105.9641 RMS: 227.7593 conf:   1.0000
INFO  [data_validator] 	val:  42.8158, lp:  42.4197 mean dev:  -6.3465 RMS:  15.1511 conf:   1.0000
INFO  [data_validator] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000

INFO  [sensors] Airspeed status:
INFO  [data_validator] 	no data

INFO  [vehicle_acceleration] selected sensor: 3670026, rate: 1000.0 Hz, estimated bias: [0.0000 0.0000 -0.1309]
INFO  [sensor_calibration] ACC 3670026 EN: 1, offset: [0.0219 -0.1406 -0.1079] scale: [1.0038 1.0172 1.0099]

INFO  [vehicle_angular_velocity] selected sensor: 3670026, rate: 8038.0 Hz FIFO
INFO  [vehicle_angular_velocity] estimated bias: [-0.0017 -0.0030 -0.0005]
[... 93 lines cut ...]
vehicle_local_position            0   19  100  1  168 
vehicle_magnetometer              0    6   43  1   40 
vehicle_magnetometer              1    4   27  1   40 
vehicle_odometry                  0    2  100  1  256 
vehicle_status                    0   33    2  1   80 
vehicle_status_flags              0    5    2  1   48 
```

Here the output for ``rostopic hz /mavros/imu/data_raw -w 100``:
``` 
[... cut ...]
average rate: 330.254                                                                                                                                         
        min: 0.001s max: 0.005s std dev: 0.00061s window: 100                                                                                                 
average rate: 325.471                                                                                                                                         
        min: 0.001s max: 0.006s std dev: 0.00068s window: 100                                                                                                 
```

dagar, 2021-05-09:
> After setting that to 400 I was able to reach up to 333Hz, but not the full 400Hz.

It's probably mavlink limiting you. 

https://github.com/PX4/PX4-Autopilot/blob/541697d193a65f75eda8d6d593a3460fe41e924d/src/modules/mavlink/mavlink_main.cpp#L2192-L2203

What does `mavlink status` and `mavlink status streams look like?

berndpfrommer, 2021-05-12:
Here's the relevant part of the output of mavlink status:
```
instance #2:
	mavlink chan: #2
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 24431.4 B/s
	  txerr: 0.0 B/s
	  tx rate mult: 0.920
	  tx rate max: 46080 B/s
	  rx: 0.0 B/s
	  rx loss: 0.0%
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: serial (/dev/ttyS2 @921600)
```
I don't understand why it shows tx rate max: 46080 B/s when I start it like this in extras.txt:
```
mavlink stream -d /dev/ttyS2 -b 921600 -m onboard -r 100000
```
Could that mavlink stream already be started elsewhere (earlier or later) such that my settings are ignored?

Anyway, here the output of ``mavlink status streams``:
```
instance #2:
	Name                Rate Config (current) [Hz] Message Size (if active) [B]
	HEARTBEAT                       1.00 (1.000)    21
	STATUSTEXT                     20.00 (18.403)   66
	COMMAND_LONG                  unlimited       
	TIMESYNC                       10.00 (9.201)    28
	CAMERA_TRIGGER                unlimited       
	HIGHRES_IMU                   400.00 (368.057)  75
```

berndpfrommer, 2021-05-12:
I was able to bump the rate in QGC to 100000, so I got this from ``mavlink status`` and ``mavlink status streams``:
```
instance #2:
	mavlink chan: #2
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 39665.7 B/s
	  txerr: 0.0 B/s
	  tx rate mult: 1.000
	  tx rate max: 100000 B/s
	  rx: 569.4 B/s
	  rx loss: 0.0%
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS2 @921600)

```
```
instance #2:
	Name                Rate Config (current) [Hz] Message Size (if active) [B]
	HEARTBEAT                       1.00 (1.000)    21
	STATUSTEXT                     20.00 (20.000)   66
	COMMAND_LONG                  unlimited       
	TIMESYNC                       10.00 (10.000)   28
	CAMERA_TRIGGER                unlimited       
	HIGHRES_IMU                   400.00 (400.000)  75

```

but I am still seeing a max of 331Hz frequency on the mavros bridge.

MichaelBosello, 2022-11-01:
I have this problem too. Is there any update on this? Did anyone find a solution?

rvannamen, 2024-01-30:
Just as @berndpfrommer I am requesting the HIGHRES_IMU data at 400 Hz, but can only receive it at 330 Hz. Has there been any proper explanation for this @julianoes?

Based on some test results I am currently suspecting that certain mavlink settings like IMU_GYRO_RATEMAX and IMU_INTEG_RATE have an effect on the maximum frequency at which the HIGHRES_IMU data can be obtained, but I can not figure out how it works exactly