PX4/PX4-Autopilot #11071: [ecl/EKF] EKF stopping navigation in every flight

glegarda, 2018-12-19:
Hello,

I am constantly getting this error in my flights: "[ecl/EKF] EKF stopping navigation". I am using a PX4FLOW and a Lidar Lite v3 for range data and I am flying without GPS. I am using fmu-v3 (master) with a small alteration to make the Lidar boot before the PX4FLOW ([https://github.com/PX4/Firmware/issues/11009](https://github.com/PX4/Firmware/issues/11009)).

From the logs I see that my local position estimates vary wildly at the beginning until I get the error and, after that, the estimating stops (obviously). However, the condition_local_position_valid flag drops to zero even before EKF stops navigation and it never goes back to one. Here is an example of a short test I did this morning:

[https://review.px4.io/plot_app?log=95d3f262-f3ef-4dea-9398-91dcce53d245](https://review.px4.io/plot_app?log=95d3f262-f3ef-4dea-9398-91dcce53d245)

I was flying outdoors and neither take-off nor landing were exactly smooth. I thought it might be an issue with the PX4FLOW, but I have tried both indoors and outdoors flights on floors with different textures and the quality data I get seems to be good enough. Also, even though the acceleration power spectral density data is not ideal, I get much better values indoors and yet I still get the same error, so I doubt vibration is an issue.

Any ideas?

julianoes, 2019-10-14:
@glegarda have you found a solution to this?

glegarda, 2019-10-14:
@julianoes unfortunately, I stopped working on this project before I could find a solution. Sorry about that.

julianoes, 2019-10-15:
Ok, thanks for following up though.

closed 2019-10-15 by julianoes