PX4/PX4-Autopilot #11073: Help for pull request

tuloski, 2018-12-19:
I added a new feature to EKF2.
The problem is that it requires both modifications to PX4/Firmware, but also to lib/ecl/EKF.
How is the correct protocol for pull request in both main firmware and submodules?
I have my fork and I can pull request for PX4/Firmware, but how to include the submodule? Do I need separate fork with separate pull request? What if the pull request is accepted in one and not in the other (the code won't compile)?

dagar, 2018-12-19:
Hi @tuloski 

Open the pull request on https://github.com/PX4/ecl, if it requires a corresponding PX4/Firmware change open a simultaneous PR with the submodule temporarily changed to your own fork. Make it clear in both PRs that they're dependent and we'll review and test them together. Once accepted we merge PX4/ecl, then immediately update the PX4/Firmware PR to point to the new ecl master and merge.

This is the worst case when changes are required on both sides. In many cases the changes can actually go in separately.

closed 2018-12-19 by dagar

tuloski, 2018-12-19:
Ok perfect I'll do that.
But before I need to fix https://github.com/PX4/Firmware/issues/11072 because idk if it is my issue in the code or a problem with QGC.