PX4/PX4-Autopilot #11074: Upload Error: px_uploader.py: No such file or directory

xrb936, 2018-12-19:
Here is the error message:

$ make intel_aerofc-v1_default upload
ninja: Entering directory `/cygdrive/c/PX4/home/Firmware/build/intel_aerofc-v1_default'
[3/4] uploading px4
Copying files to Aero board (root@10.88.193.202)...
The authenticity of host '10.88.193.202 (10.88.193.202)' can't be established.
ECDSA key fingerprint is SHA256:xMmdPK4036rjoNP2mFfKF5j3KhO4QF8An2QnfAUY9s4.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '10.88.193.202' (ECDSA) to the list of known hosts.
root@10.88.193.202's password:
intel_aerofc-v1_default.elf                                                           100%   13MB  20.6MB/s   00:00
/cygdrive/c/PX4/home/Firmware/boards/intel/aerofc-v1/scripts/px_uploader.py: No such file or directory
FAILED: boards/intel/aerofc-v1/CMakeFiles/upload
cd /cygdrive/c/PX4/home/Firmware/build/intel_aerofc-v1_default/boards/intel/aerofc-v1 && /cygdrive/c/PX4/home/Firmware/boards/intel/aerofc-v1/scripts/aero_upload.sh /cygdrive/c/PX4/home/Firmware/build/intel_aerofc-v1_default/intel_aerofc-v1_default.elf
ninja: build stopped: subcommand failed.
make: *** [Makefile:171: intel_aerofc-v1_default] Error 1

I have tried several times on different computer, same issue.

mhkabir, 2018-12-19:
Looks like a regression after the board restructure.

dagar, 2018-12-19:
The boards are able to provide their own custom upload now. https://github.com/PX4/Firmware/blob/master/boards/intel/aerofc-v1/CMakeLists.txt

I can review this on an aero this weekend.

dagar, 2018-12-19:
Looks like the aero upload script needs to know the path to px_uploader.py under Tools/ and not under the boards scripts directory.

https://github.com/PX4/Firmware/blob/master/boards/intel/aerofc-v1/scripts/aero_upload.sh#L43

xrb936, 2018-12-19:
@dagar Thanks, let me try the new script. Should I re-clone and re-make it?

xrb936, 2018-12-19:
I moved the py file to that folder, and now I have this issus:

/dev/ttyS1:
Updating firmware on AeroFC
Traceback (most recent call last):
  File "/root/px_uploader.py", line 880, in <module>
    main()
  File "/root/px_uploader.py", line 736, in main
    fw = firmware(args.firmware)
  File "/root/px_uploader.py", line 122, in __init__
    self.desc = json.load(f)
  File "/usr/lib/python2.7/json/__init__.py", line 291, in load
    **kw)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
Firmware updated
Restarting router
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Finished.

trexgerratt, 2019-06-19:
I have the same issue with the "ValueError: No JSON object could be decoded" Any progress here?

trexgerratt, 2019-06-19:
I figured it out. In my case, I was trying to flash the ".elf" file when it should have been the ".px4" file from the make command.

Hope that helps others!

caiespin, 2019-07-11:
Hi @trexgerratt How did you managed to select the ".px4" binary instead of the ".elf"?

julianoes, 2019-07-15:
I think this has been resolved in latest master, right?

skamano, 2019-07-22:
I just cloned master and tried uploading it to my Intel drone and I've been having the exact same issue as @trexgerratt had. So far I haven't found a fix, but I'll try specifying the .px4 file for px_uploader.py

julianoes, 2019-07-25:
@skamano can you check #11075?