PX4/PX4-Autopilot #11076: missing terminating " character [-Werror](build_git_version.h)

473867143, 2018-12-20:
Platform：VMware Ubuntu16.04
PX4 version:v1.8.0
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) 
arm-none-eabi-g++:gcc version 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715] (GNU Tools for ARM Embedded Processors) 

when i run  **make px4fmu-v3_default** failed

Here is the error message:

[59/1051] Building C object src/lib/ve...n/CMakeFiles/version.dir/version.c.obj
FAILED: /home/px4/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc  -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V2 -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -Isrc/lib/version -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -std=gnu99 -fno-common -Wall -Wextra -Werror -Warray-bounds -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wbad-function-cast -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG -MMD -MT src/lib/version/CMakeFiles/version.dir/version.c.obj -MF src/lib/version/CMakeFiles/version.dir/version.c.obj.d -o src/lib/version/CMakeFiles/version.dir/version.c.obj   -c ../../src/lib/version/version.c
In file included from ../../src/lib/version/version.c:36:0:
src/lib/version/build_git_version.h:6:30: error: missing terminating " character [-Werror]
 #define PX4_GIT_VERSION_STR  "11:30:13.203576 git.c:344               trace: built-in: git 'rev-parse' '--verify' 'HEAD'
                              ^
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
[59/1051] Building CXX object src/modu...transport/uc_transfer_receiver.cpp.obj
ninja: build stopped: subcommand failed.
Makefile:154: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1

**When I compile on Qt5.11.0,it's succeed.**
-- PX4 VERSION: v1.8.0-1253-ga1af7ff
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- C compiler: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
-- C++ compiler: g++ (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-590-g2354c30
-- Configuring done
-- Generating done

**but run failed.you can see :** 
Starting /home/px4/src-1.8.0/build-Firmware-Desktop_Qt_5_11_0_GCC_64bit-Default/bin/px4...
INFO  [Unknown] assuming working directory is rootfs, no symlinks needed.
ERROR [Unknown] Error opening startup file, does not exist: etc/init.d/rcS
/home/px4/src-1.8.0/build-Firmware-Desktop_Qt_5_11_0_GCC_64bit-Default/bin/px4 exited with code 255

dagar, 2018-12-20:
@473867143 can you reproduce the failure and share the output of `build/posix_sitl_default/build_git_version.h`?

473867143, 2018-12-21:
@dagar 
The problem may be that I changed the source code of git myself and reinstalled the day before yesterday. I just want to solve the problem with git clone error: RPC failed; curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated. Then compile has problems similar to git version
this is the failure.

/* Auto Magically Generated file */
/* Do not edit! */
#pragma once
#define PX4_GIT_VERSION_STR  "f160743df40b9dcc9558f3df1c6eeaf56646a758"
#define PX4_GIT_VERSION_BINARY 0xf160743df40b9dcc
#define PX4_GIT_TAG_STR  "v1.6.5-dirty"
#define PX4_GIT_BRANCH_NAME  ""

#define NUTTX_GIT_VERSION_STR  "8b81cf5c7ece0c228eaaea3e9d8e667fc4d21a06"
#define NUTTX_GIT_VERSION_BINARY 0x8b81cf5c7ece0c22
#define NUTTX_GIT_TAG_STR  "upstream_7.18+"

#define MAVLINK_LIB_GIT_VERSION_STR  "b69eccedfcbec1f64a0340afa0c2294ca3559cfb"
#define MAVLINK_LIB_GIT_VERSION_BINARY 0xb69eccedfcbec1f6

473867143, 2018-12-24:
`~/src-1.6.5/Firmware/build_px4fmu-v2_default/build_git_version.h`

/* Auto Magically Generated file */
/* Do not edit! */
#pragma once
#define PX4_GIT_VERSION_STR  "10:32:53.292363 git.c:344               trace: built-in: git 'rev-parse' '--verify' 'HEAD'
f160743df40b9dcc9558f3df1c6eeaf56646a758"
#define PX4_GIT_VERSION_BINARY 0x10:32:53.292363 
#define PX4_GIT_TAG_STR  "10:32:49.572565 git.c:344               trace: built-in: git 'describe' '--always' '--tags' '--dirty'
10:32:50.081732 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:50.082418 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:50.084074 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:50.085109 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:50.085478 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:50.086913 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:50.258754 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:50.259123 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:50.260521 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:51.195221 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:51.195722 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:51.206157 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:51.479297 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:51.480185 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:51.484097 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.384849 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.385432 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.386909 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.407687 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.408409 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.409818 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.423787 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.424411 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.425866 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.446838 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.447578 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.449044 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.575549 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.576746 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.580163 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.669181 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.670006 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.673798 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.676726 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.677697 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.681787 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:52.829424 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:52.839814 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:52.843184 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:53.182876 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:53.183441 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:53.184958 git.c:344               trace: built-in: git 'status' '--porcelain'
10:32:53.216026 run-command.c:334       trace: run_command: 'status' '--porcelain'
10:32:53.217145 exec_cmd.c:120          trace: exec: 'git' 'status' '--porcelain'
10:32:53.219771 git.c:344               trace: built-in: git 'status' '--porcelain'
v1.6.5-dirty"
#define PX4_GIT_BRANCH_NAME  ""

#define NUTTX_GIT_VERSION_STR  "10:32:54.810709 git.c:344               trace: built-in: git 'rev-parse' '--verify' 'HEAD'
8b81cf5c7ece0c228eaaea3e9d8e667fc4d21a06"
#define NUTTX_GIT_VERSION_BINARY 0x10:32:54.810709 
#define NUTTX_GIT_TAG_STR  "10:32:53.524929 git.c:344               trace: built.0
upstream_7.18+"

#define MAVLINK_LIB_GIT_VERSION_STR  "10:32:54.820191 git.c:344               trace: built-in: git 'rev-parse' '--verify' 'HEAD'
b69eccedfcbec1f64a0340afa0c2294ca3559cfb"
#define MAVLINK_LIB_GIT_VERSION_BINARY 0x10:32:54.820191

karliss, 2019-01-04:
Does [px_update_git_header.py](/PX4/Firmware/blob/master/src/lib/version/px_update_git_header.py) really needs to capture stderr? In most cases the output we are interested in will be in stdout and if there is any extra output on stderr it will produce broken build_git_version.h. We don't have to worry about critical git command errors as those should cause non zero return code and subprocess to throw an exception.  I suggest removing most of the stderr arguments and keeping default check_output behaviour which prints stderr to stderr.  I guess that in on instance it was really necesarry to capture stderr (maybe tests) and the rest are bad copy paste.

closed 2019-07-08 by stale[bot]