PX4/PX4-Autopilot #11082: RunCam Split Mini in px4?

manotroll, 2018-12-21:
is it possible to use the RunCam Split Mini in px4?
How do I make her serial connection on px4?
it can be shared with the miniosd since runcan would practically not receive external commands 
How do other open-source controllers already support RunCam not add this support to px4?

mhkabir, 2018-12-21:
Please use discuss.px4.io for such questions.

mhkabir, 2018-12-21:
From a first pass, it seems it would not be hard to add a driver for it.

closed 2018-12-21 by mhkabir

manotroll, 2018-12-21:
I do not know how to program
so how would it help?

manotroll, 2018-12-22:
RunCam Support (Customer Services)

Dec 22, 10:08 CST
Sorry for that Split mini  doesn't support PWM control. 

Regards,
Stella T.
RunCam Customer Service

as it does not support pwm control would not it be possible to add its support?
if it was compatible many would use it I already think that the controllers that supports px4 have more serial than others