PX4/PX4-Autopilot #11085: serial port test

manotroll, 2018-12-21:
how can I test the serial ports of px4 in a simple way?
because m8n is working on the pc and is not on px4

manotroll, 2018-12-25:
I bought another module because m6n is working although m8n funsione in the computer

dlarue, 2019-07-08:
That's one way for it to go away....

julianoes, 2019-07-10:
Sorry for the late response here.

How do you check "is working"? Can you do `gps status` in Mavlink shell to check what's up?

And when you connect to the GPS via computer, can you use the uBlox tool to check the settings of the GPS? Maybe some setting is incompatible with what the PX4 driver tries to do.

manotroll, 2019-07-10:
so I bought another GPS and solved it but the same GPS still works on the pc

dlarue, 2019-07-10:
what got me was how the GPS i had connected worked in Betaflight but would not work in PX4... This was on an omnibus

manotroll, 2019-07-13:
But there's another problem with the

closed 2019-07-13 by manotroll