PX4/PX4-Autopilot #11090: New simulator TCP link do not work with Windows Cygwin Toolchain

bys1123, 2018-12-22:
I saw @julianoes update the simulator_mavlink.cpp file. Thanks for your work.

And I tried in Cygwin toolchain, seems not work well.

error:
```bash
WARN  [simulator] Failed sending mavlink message: Address family not supported by protocol
```

[image]

bys1123, 2018-12-22:
I think this bug is around here:

[simulator_mavlink.cpp](https://github.com/PX4/Firmware/blob/a1fc44d1f03969614a22019ab61af61848a271d7/src/modules/simulator/simulator_mavlink.cpp)
start line 654:
```
	struct sockaddr_in _myaddr {};
	_myaddr.sin_family = AF_INET;
	_myaddr.sin_addr.s_addr = htonl(INADDR_ANY);
	_myaddr.sin_port = htons(port);
```

MaEtUgR, 2018-12-23:
When I revert simulator and jMAVsim back to UDP it's even worse. I don't see the `Address faily not supported error` anymore but that's probably because the simulator is hanging. At least when I try to use `uorb top` everything freezes completely.

TSC21, 2018-12-23:
It can be related with the lockstep instead. @julianoes

julianoes, 2018-12-26:
I think I had fixed this and maybe stashed it away. It could be that we need to use `send(` instead of `sendto` on Win. Let me check.

PR #11101 (2018-12-27): simulator: fix TCP on Cygwin-Windows and updated two world files for Gazebo
It turns out that `sendto` does not work for TCP on Cygwin-Windows, instead we need to use `send`. This required some refactoring since we need to have the internet protocol and the port stored as a member
variable.

This should fix that lockstep SITL simulation was not working on Windows.

Fixes #11090.

I've also updated sitl_gazebo since that fixes a worldfile which needed updating for lockstep.
closed 2018-12-27 by julianoes