PX4/PX4-Autopilot #11100: ERROR [px4] Error opening startup file, does not exist: etc/init.d/rcS

karanmartian, 2018-12-26:
Hi,

I am getting this error when trying to launch any of the launch files. Please help!

Following is the log:

karan@karan-LinuxTP:~/catkin_ws/src/PX4/Firmware$ roslaunch px4 posix_sitl.launch
... logging to /home/karan/.ros/log/e2002708-08e6-11e9-9404-a434d94586cf/roslaunch-karan-LinuxTP-13574.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://karan-LinuxTP:44787/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    sitl (px4/px4)
    vehicle_spawn_karan_LinuxTP_13574_6725904797963555098 (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[sitl-1]: started with pid [13589]
INFO  [px4] Creating symlink /home/karan/catkin_ws/src/PX4/Firmware -> /home/karan/.ros/etc
ERROR [px4] Error opening startup file, does not exist: etc/init.d/rcS
process[gazebo-2]: started with pid [13590]
process[gazebo_gui-3]: started with pid [13599]
process[vehicle_spawn_karan_LinuxTP_13574_6725904797963555098-4]: started with pid [13604]
================================================================================REQUIRED process [sitl-1] has died!
process has died [pid 13589, exit code 255, cmd /home/karan/catkin_ws/devel/lib/px4/px4 /home/karan/catkin_ws/src/PX4/Firmware /home/karan/catkin_ws/src/PX4/Firmware/posix-configs/SITL/init/ekf2/iris __name:=sitl __log:=/home/karan/.ros/log/e2002708-08e6-11e9-9404-a434d94586cf/sitl-1.log].
log file: /home/karan/.ros/log/e2002708-08e6-11e9-9404-a434d94586cf/sitl-1*.log
Initiating shutdown!
================================================================================
[vehicle_spawn_karan_LinuxTP_13574_6725904797963555098-4] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/gazebo_ros/spawn_model", line 20, in <module>
    import rospy
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/melodic/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/std_msgs/msg/_Bool.py", line 5, in <module>
    import genpy
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 8, in <module>
    from loader import *
  File "/usr/lib/python2.7/dist-packages/yaml/loader.py", line 9, in <module>
    from resolver import *
  File "/usr/lib/python2.7/dist-packages/yaml/resolver.py", line 183, in <module>
[gazebo_gui-3] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/catkin_find", line 3, in <module>
    from __future__ import print_function
KeyboardInterrupt
[gazebo-2] killing on exit
[sitl-1] killing on exit
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/catkin_find", line 11, in <module>
    from catkin.find_in_workspaces import find_in_workspaces
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/find_in_workspaces.py", line 36, in <module>
    from catkin_pkg.packages import find_packages
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 38, in <module>
    from .package import _get_package_xml
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/package.py", line 41, in <module>
    import xml.dom.minidom as dom
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 22, in <module>
    from xml.dom.xmlbuilder import DOMImplementationLS, DocumentLS
  File "/usr/lib/python2.7/xml/dom/xmlbuilder.py", line 1, in <module>
    |\.(?:nan|NaN|NAN))$''', re.X),
  File "/usr/lib/python2.7/re.py", line 194, in compile
    return _compile(pattern, flags)
  File "/usr/lib/python2.7/re.py", line 249, in _compile
    p = sre_compile.compile(pattern, flags)
KeyboardInterrupt
    """Implementation of the DOM Level 3 'LS-Load' feature."""
KeyboardInterrupt
shutting down processing monitor...
... shutting down processing monitor complete
done

julianoes, 2018-12-26:
@karanmartian I've added ``` around your pasted output so it's easier to read.

Have you done all sourcing steps described here?
https://dev.px4.io/en/simulation/ros_interface.html#launching-gazebo-with-ros-wrappers

karanmartian, 2018-12-26:
thnx @julianoes, so yea the stable releases need to be updated to include the init.d-posix folder without which the roslaunch launch files will issue error since they will not find the correct rcS

julianoes, 2018-12-26:
> so yea the stable releases need to be updated to include the init.d-posix folder without which the roslaunch launch files will issue error since they will not find the correct rcS

Is this a question or a statement? And does that mean your problem is fixed?

karanmartian, 2018-12-26:
This is a bug that needs to be fixed in the repo I would say, but you can say that my problem is fixed but maybe not for others.

julianoes, 2018-12-26:
Ok, thanks. Can you share a snippet or diff how you fixed this?

karanmartian, 2018-12-26:
@TSC21 if you clone the repo freshly, do a "git checkout v1.8.2" and do a roslaunch on the launch files does it work out of the box? Thats what I am trying to address

TSC21, 2018-12-29:
> @TSC21 if you clone the repo freshly, do a "git checkout v1.8.2" and do a roslaunch on the launch files does it work out of the box? Thats what I am trying to address

It won't work out of the box with 1.8.2. As said, this has been fixed on upstream master.

closed 2018-12-29 by TSC21