PX4/PX4-Autopilot #11114: Snapdragon start instructions don't work

julianoes, 2018-12-29:
I could not figure out (anymore) how to actually start px4 on Snapdragon.
It's not clear reading through https://docs.px4.io/en/flight_controller/snapdragon_flight.html.

I tried the old command (see below) but it doesn't work anymore.
Any chance you could help @karliss? Thanks.

```
root@linaro-developer:/# cd home/linaro/
root@linaro-developer:/home/linaro# ./px4 mainapp.config
INFO  [px4] Creating symlink mainapp.config -> /home/linaro/etc
ERROR [px4] Error opening startup file, does not exist: etc/init.d/rcS
```

karliss, 2018-12-29:
Try "px4 -s mainapp.config".

I observed that some of the scripts copy second config file (px4.config) to adsp folder. From the content of it seems like it should be executed on qurt side. I couldn't get that working. So for starting qurt modules I am using qshell command from main config similar to https://github.com/PX4/Firmware/blob/master/posix-configs/eagle/init/rcS .  [This comments](https://github.com/PX4/Firmware/blob/ffffcaeb0f4ac1c512201e8dc784d528b04cdcfe/platforms/qurt/src/px4_layer/main.cpp#L185) also indicates that this might be the current intended behavior.

julianoes, 2018-12-29:
Ok turns out the command is:
```
./px4 -s mainapp.config rootfs/
```
It's just not documented anywhere.

karliss, 2018-12-29:
It should match the common posix code used by other linux based platfroms.

dagar, 2018-12-29:
I also find it quite confusing. My personal preference would be to push everything into the binary itself. This makes debugging and deployment easier/obvious.

julianoes, 2018-12-29:
@dagar so you would bake the config files into the binary?

julianoes, 2018-12-29:
The problem is too that we don't have a generic way to have more than one autostart config for targets like Snapdragon or RPi.

dagar, 2018-12-29:
> @dagar so you would bake the config files into the binary?

Yes, then at startup we respect the regular SYS_AUTOSTART param. There's also no reason the external script mechanism couldn't be preserved if someone really wanted it.

> The problem is too that we don't have a generic way to have more than one autostart config for targets like Snapdragon or RPi.

On Linux we're getting there. I want the vast majority of init to be unified across platforms. We simply miss too many trivial problems with this kind of code sprawl.

julianoes, 2018-12-29:
I agree but I feel like config files like these should be actual files (and pushed like that) just because it's clearer and less magic. However, I'm all for unifying it all.

dagar, 2018-12-29:
What's needed (either way) is a proper package for deployment. PX4 needs to be installed somewhere on the system, including integration with the init system. Dumping all the files to a home directory and starting out of rc.local isn't great. 

In the snapdragon case the package would include both the linux binary, startup config, dsp dynamic library, and probably setup the area for parameter store and logging.

We'd probably use systemd to manage the daemon.

julianoes, 2018-12-29:
Yes, that all sounds reasonable. And in that case all the startup files should probably go either into `/usr/var/` or similar.

julianoes, 2019-07-01:
Closing for now, can be improved if we get more support from people behind Snappy again.

closed 2019-07-01 by julianoes