PX4/PX4-Autopilot #11116: Poor Fixed Wing flight performance in v1.9-alpha (SITL)

Antiheavy, 2018-12-29:
I was doing some SITL testing on recent Master and v1.9-alpha and noticed some problems with Fixed Wing:

1) Ground track often is parallel to the mission track instead of on it, sometimes by quite a lot.
2) Poor loiter entry and circle following (worse than normal)
3) Tangent and heading requirements for loiter exit is being ignored.
4) Loiter orbits are off-center from the proper center point
5) super aggressive pitch response to things like Abort or other abrupt altitude changes.

See this QGC screenshot from SITL:
[image]

dagar, 2019-01-04:
I suspect at least a portion of this can be blamed on the SITL plane model itself.

Antiheavy, 2019-01-04:
> I suspect at least a portion of this can be blamed on the SITL plane model itself.

I'm not so sure.  Not not respecting tangent loiter exit and the parallel mission tracks are definitely some logic problem somewhere.  You can get it to follow the landing approach parallel several hundred meters to the side depending on which way it crazy-ivan turns out of the loiter.

ryanjAA, 2019-01-05:
Might be time to get the foam planes out and see what happens.

Antiheavy, 2019-01-05:
> Might be time to get the foam planes out and see what happens.

I'll try re-installing SITL this weekend and seeing if I can repeat the issues from scratch.

Antiheavy, 2019-01-05:
Here are two screenshots of identical mission plans in QGC 3.4.4.  The left is with PX4 v1.8.0 (trivially modified) and the right is from PX4 Master.  I'm not familiar with the SITL changes from POSIX_SITL to the new PX4_SITL, either way the issues listed above seem like bugs or problems:
[image]

LorenzMeier, 2019-01-05:
Thanks, that's super helpful!

CHschizo, 2019-01-09:
The line tracking offset is due to the simulated wind. I experienced the same problem with the simulation. In your simulation you have wind in East-direction which will not be corrected by the plane. I don't know why. But if you change wind direction you will get the offset in the corresponding direction. I found out that in the simulation, PX4 is not able to estimate the wind direction and assumed that this is why in the simulation the offset in path-following persists.

Antiheavy, 2019-01-09:
>  I found out that in the simulation, PX4 is not able to estimate the wind direction

interesting.  I had wondered if it was wind, but I did not see any indications in the telemetry stream of wind effects.  the airspeed and ground speed where typically reporting nearly identical.  Maybe I'll try again.  I'm only using the default PX4 Fixed Wing Gazebo settings.

Wind would not explain the changes in the loiter entry and exit performance that I am seeing.

ryanjAA, 2019-01-11:
If there’s no more news on this by next week or so, I’ll get someone on our flight test team to get a small foam plane (bixler, etc) up in the air and get a log uploaded to look at and we can see actually what’s happening.

Antiheavy, 2019-01-27:
Here is some more SITL testing showing additional problems with Fixed Wing SITL flight performance.  I'd like to break these into separate individual issues if someone can help me determine more specifically what might be going on, it all seems somewhat related...

Additional issues:
- Fixed Wing Landing Abort does not always go to the same loiter altitude above the Land WP.
- Fixed Wing Landing Abort turns hard off the landing ground track and takes a wide turn towards the loiter above the Land WP.

[image]

Antiheavy, 2019-01-28:
Update:
I finally figured out how to set the Wind Force to 0 in Gazebo SITL (thanks @TSC21 ).  @CHschizo is correct, the SITL seems to not correctly deal with the simulated wind.

With wind = 0 the SITL waypoint tracking problems went away.  Line following is correct, loiter orbits are centered correctly, and landing aborts track correctly relative straight ahead.

[image]

So this turns into 2 different issues for which I will open new github issue posts:
1) there still seems to be a bug with tanget loiter exit
2) there is some bug(s) with wind in SITL.

closed 2019-01-28 by Antiheavy