PX4/PX4-Autopilot #11118: Commander spams the nsh console on SITL

TSC21, 2018-12-31:
**Describe the bug**
On the process of testing the micro RTPS bridge between a SITL client (with Gazebo) and the agent, I get the following warning msgs on the output of the `nsh` console:
```sh
WARN  [px4] commander: px4_sem_timedwait failure: ret: 110, Connection timed out
WARN  [cdev] commander: px4_poll() sem error: Success
WARN  [px4] commander: px4_sem_timedwait failure: ret: 110, Connection timed out
WARN  [cdev] commander: px4_poll() sem error: Success
```
which repeat themselves over time. The same does not happen if one builds the posix build using `_default` instead of `_rtps`.

I am testing latest upstream master. This may have been introduced by the latest changes on SITL and with the lockstep feature.

I am currently trying to find if this is just a warning or if this is introducing some sort of problem or breaking some functionality. Though, this wasn't happening before.

@julianoes @MaEtUgR any tips on what's causing this?

**To Reproduce**
1. Clone upstream master;
2. Issue `make px4_sitl_rtps gazebo`
3. Observe the above lines on the `nsh` console output.

**Expected behavior**
A clean console output.

**Log Files and Screenshots**
[image]

julianoes, 2019-01-01:
Hm, two things are odd:
1. Why does px4_poll time out?
2. Why does it say `sem error: Success`.

I think the first one we need to figure out for `_rtps` and the second one is probably just unclear user feedback.

LorenzMeier, 2019-01-01:
Couple notes: Both happens when handling `ETIMEDOUT`. The definition of this is:

`#define ETIMEDOUT       110     /* Connection timed out */`

I would look for sign errors (usage of 110 vs. -110) in the handling of the error. In particular around this section here:
https://github.com/PX4/Firmware/blob/73ed9deac507c86892afadee5bca2024b5b8da17/platforms/posix/src/px4_layer/px4_sem.cpp#L143-L152

LorenzMeier, 2019-01-01:
If you look at the above tracing the code would suggest that you can't have the `printf()` with error 110 as it is explicitly ruled out.

LorenzMeier, 2019-01-01:
Commit to blame is 98ae0186e93 - I don't know why the `errno` change slipped through in the review, but it seems an unnecessary change.

PR #11122 (2019-01-01): Fix usage of errno in POSIX semaphore error handling
Previous commit 98ae0186e93 changed the err variable to errno, which is not the return value we are looking for.

Fixes #11118.
closed 2019-01-02 by TSC21