PX4/PX4-Autopilot #11119: Compilation in cygwin unable to find external include path mentioned in cmakelist

Ankur1014, 2018-12-31:
**Describe the bug**
I am adding my own module in px4 firmware folder. The header files for my modules source code are placed in an external folder (outside px4 firmware directory) at *D:/Project/PX4/include*. I was adding the required paths for my module in the Cmakelist.txt as follows

`include_directories(
"/mnt/d/Project/PX4/include"
)` 

Till now I was using WSL on windows for compilation and it was working fine. Now I am trying to switch to cygwin for compilation in Windows. 
I modified the cmakelist for cygwin as follows:

`include_directories(
"/cygdrive/d/Project/PX4/include"
)` 

However, the compilation is failing for the Cmakelist.txt complaining that is unable to find any of the external header files.

`FAILED: src/modules/px4_simulink_app/CMakeFiles/modules__px4_simulink_app.dir/nuttxinitialize.cpp.obj
/cygdrive/c/px4_cygwin/toolchain/gcc-arm/bin/arm-none-eabi-g++.exe    -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include 
-I/cygdrive/d/Project/PX4/include
-fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization  -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fpermissive -Wno-narrowing -Os -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Wframe-larger-than=2000 -MD -MT src/modules/px4_app/CMakeFiles/modules__px4_app.dir/nuttxinitialize.cpp.obj -MF src/modules/px4_app/CMakeFiles/modules__px4_app.dir/nuttxinitialize.cpp.obj.d -o src/modules/px4_app/CMakeFiles/modules__px4_app.dir/nuttxinitialize.cpp.obj -c ../../src/modules/px4_app/nuttxinitialize.cpp
../../src/modules/px4_app/nuttxinitialize.cpp:11:10: fatal error: nuttxinitialize.h: No such file or directory
 #include "nuttxinitialize.h"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.`

**To Reproduce**
On Windows, clone PX4 v1.8.0 firmware.
Move the header files for any module ouside firmware directory.
Modify the cmakelist accodringly. 
Try to build in Cygwin.

 
**Full Cmakelist**

Below is my full CMakelist.txt

include_directories(
"/cygdrive/d/Project/PX4/include"
)

px4_add_module(
MODULE modules__px4_app
MAIN px4_simulink_app
STACK_MAIN 2000
SRCS
   cygwinTest1.cpp
   cygwinTest1_data.cpp
   nuttxinitialize.cpp
DEPENDS
    platforms__common
)

string (REPLACE "-Werror" "" CMAKE_C_FLAGS "${CMAKE_C_FLAGS}")
string (REPLACE "-Werror" "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fpermissive -Wno-narrowing")

Any help in resolving this issue is much appreciated

Ankur1014, 2019-01-02:
@MaEtUgR , can you please help me with this issue?

closed 2019-01-02 by mhkabir

MaEtUgR, 2019-01-06:
@Ankur1014 Generally this should work maybe something with the paths is handeled a bit differently and exposes a problem with the includes in your particular case. Did you already solve it? I don't see the complete overview of paths in your description.

`-I/cygdrive/d/Project/PX4/include` is there like you configured and you should be able to include the header directly in this directory with e.g. `#include <nuttxinitialize.h>`. But I presume the headerfile is somewhere in a lower level because given the cpp file path `../../src/modules/px4_app/nuttxinitialize.cpp`.

How is the path structure now? Did you mind the `<>`/`""` include rules?
Sorry for searching in the dark I've never tried this before.