PX4/PX4-Autopilot #11121: Inaccurate Voltage-based Estimation Fused with Current Integration

alessio-zatti, 2019-01-01:
Hi Team,
I'm working with PIXHAWK4 equipped with the Power Module PM07.

Battery: 3S 5000mAh 50C (I always used full charged battery) - 3mOhm the internal R
Frame: Quadcopter FJ450
Motors: 2312E DJI
ESC: BLHeli 30A

Trying to configure and tune the battery module, I found issues with the Voltage-based Estimation  method.
I configured all the parameters according to the guide https://docs.px4.io/en/config/battery.html but the battery remains percentage decrease quickly (from 100% to 30% in no more than two minutes). Checking the battery status at the end of my test this discharging ramp is not correct.

I configured the current compensation method and the percentage is more reliable.

You can check the  log:

USING THE VOLTAGE-BASED METHOD:
[https://logs.px4.io/plot_app?log=1e248411-e512-40ec-9bda-e6f5867728d7](url)

USING THE CURRENT COMPENSATION:
[https://logs.px4.io/plot_app?log=6b86e36c-4521-43e9-83f6-32088ba9a01d](url)

What I'm doing wrong?

I checked the code and I found something strange.  It seems the first value of _remaining is discarded. Im referring to this method:

`
void
Battery::estimateRemaining(float voltage_v, float current_a, float throttle, bool armed)
{	// remaining battery capacity based on voltage
	float cell_voltage = voltage_v / _n_cells.get();

	// correct battery voltage locally for load drop to avoid estimation fluctuations
	if (_r_internal.get() >= 0.f) {
		cell_voltage += _r_internal.get() * current_a;

	} else {
		// assume linear relation between throttle and voltage drop
		cell_voltage += throttle * _v_load_drop.get();
	}

	_remaining_voltage = math::gradual(cell_voltage, _v_empty.get(), _v_charged.get(), 0.f, 1.f);

	// choose which quantity we're using for final reporting
	if (_capacity.get() > 0.f) {
		// if battery capacity is known, fuse voltage measurement with used capacity
		if (!_battery_initialized) {
			// initialization of the estimation state
			_remaining = _remaining_voltage;

		} else {
			// The lower the voltage the more adjust the estimate with it to avoid deep discharge
			const float weight_v = 3e-4f * (1 - _remaining_voltage);
			_remaining = (1 - weight_v) * _remaining + weight_v * _remaining_voltage;
			// directly apply current capacity slope calculated using current
			_remaining -= _discharged_mah_loop / _capacity.get();
			_remaining = math::max(_remaining, 0.f);
		}

	} else {
		// else use voltage
		_remaining = _remaining_voltage;
	}
}
`
I'm referring to this part:

`
else {
		// The lower the voltage the more adjust the estimate with it to avoid deep discharge
		const float weight_v = 3e-4f * (1 - _remaining_voltage);
		_remaining = (1 - weight_v) * _remaining + weight_v * _remaining_voltage;
		// directly apply current capacity slope calculated using current
		_remaining -= _discharged_mah_loop / _capacity.get();
		_remaining = math::max(_remaining, 0.f);
	}
`

Thanks
Alessio

hamishwillee, 2019-01-01:
@MaEtUgR Might be something you can help with?

alessio-zatti, 2019-01-02:
Hi Hamish,
I had some trouble configuring the battery. To compare the two methods ("Current-based load compensation" and "voltage-based estimate merged with the current integration") I fixed the drone on the ground and I used a tester to measure the current. With the throttle stick low the value shown by QGroundControl and the value measured by the tester are very different, but when I moved the stick to the center (in hovering if the drone was free to fly) the two values are equal (~8A) but in one case, that one with the parameter BAT_CAPACITY equal to 4500 ("voltage-based estimate merged with the current integration") the capacity percentage decreased very quickly. 
I do not undestand this behaviour.

Thanks

hamishwillee, 2019-01-02:
@alessio-zatti Thanks for the information. I am not particularly familiar with this either, but @MaEtUgR is, and should be able to comment when he returns from his summer break.

MaEtUgR, 2019-01-06:
@alessio-zatti Sorry to hear about your problems. I replaced your code citations with GitHub references to the current implementations. Let me have a look at the log. I didn't spot the cause yet.

> I checked the code and I found something strange. It seems the first value of _remaining is discarded.

The battery estimation values are initialized to default values on boot because the potentially unavailable hardware reporting the measurements is not initialized at the time. When the first useful value is received the initialization is redone based on the first voltage measured because otherwise the filter would very slowly converge towards the correct battery state.

The else block you referenced is a complementary filter that fuses voltage and current based estimate and adjusts the weight towards the end of the battery's life because when the voltage drops too low that's a known hard limit and trust on current based measurement is reduced.

Generally if the voltage based method doesn't work properly it needs to be correctly configured first. The advantage of current based estimation is foremost a more stable estimation under dynamic load and a more accurate estimation when starting with a full battery.

closed 2019-07-08 by stale[bot]