PX4/PX4-Autopilot #11125: mavlink 0 (USB) abnormal cpu usage

dagar, 2019-01-01:
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
	   0 Idle Task                  833040 20.312   556/  748   0 (  0)  READY  3
	   1 hpwork                     202538  6.324   768/ 1780 249 (249)  READY 18
	   2 lpwork                       4301  0.148   720/ 1780  50 ( 50)  w:sig  8
	   3 init                         3945  0.000  2000/ 2484 100 (100)  w:sem  3
	 522 top                           281  4.166  1304/ 1684 255 (255)  RUN    3
	  11 dataman                         1  0.000   728/ 1180  90 ( 90)  w:sem  4
	 518 log_writer_file                 0  0.000   376/ 1148  60 ( 60)  w:sem 33
	 149 sensors                    230883  7.291  1304/ 1964 249 (249)  READY 21
	 151 commander                   35981  1.116  1528/ 3212 140 (140)  w:sig 31
	 152 commander_low_prio            105  0.000   616/ 2996  50 ( 50)  w:sem 31
	 158 mavlink_if0                857303 24.479  1616/ 2532 100 (100)  READY 34
	 159 mavlink_rcv_if0             15035  0.446  1120/ 2836 175 (175)  w:sem 34
	 228 gps                          3788  0.148  1064/ 1444 220 (220)  w:sem  5
	 252 mavlink_if1                116548  3.645  1528/ 2484 100 (100)  READY 22
	 253 mavlink_rcv_if1             17152  0.520  1144/ 2836 175 (175)  w:sem 22
	 268 mavlink_if2                146031  4.464  1560/ 2492 100 (100)  w:sig 26
	 269 mavlink_rcv_if2             17288  0.446  1120/ 2836 175 (175)  w:sem 26
	 283 px4io                      132266  3.869   880/ 1484 252 (252)  w:sem 11
	 463 ekf2                       395388 12.500  4536/ 6572 250 (250)  w:sem 20
	 465 mc_att_control             163645  5.357   904/ 1660 251 (251)  w:sem 19
	 468 mc_pos_control              33787  1.116   896/ 1860 250 (250)  w:sem 11
	 474 navigator                    4471  0.148   912/ 1764 105 (105)  w:sem 16
	 498 logger                      36669  1.041  1288/ 3540 245 (245)  w:sem 33
	 517 vmount                       3912  0.148   824/ 1868 100 (100)  w:sem  7

	Processes: 24 total, 6 running, 18 sleeping, max FDs: 54
	CPU usage: 77.38% tasks, 2.31% sched, 20.31% idle
	DMA Memory: 5120 total, 1024 used 1024 peak
	Uptime: 3297.552s total, 833.040s idle
	nsh>

dagar, 2019-01-01:
nsh> mavlink status

	instance #0:
		mavlink chan: #0
		type:           USB CDC
		flow control: OFF
		rates:
		  tx: 0.000 kB/s
		  txerr: 0.696 kB/s
		  tx rate mult: 0.050
		  tx rate max: 800000 B/s
		  rx: 0.000 kB/s
		FTP enabled: YES, TX enabled: YES
		mode: Config
		MAVLink version: 1
		transport protocol: serial (/dev/ttyACM0 @2000000)

	instance #1:
		mavlink chan: #1
		type:           GENERIC LINK OR RADIO
		flow control: OFF
		rates:
		  tx: 0.403 kB/s
		  txerr: 0.000 kB/s
		  tx rate mult: 1.000
		  tx rate max: 800 B/s
		  rx: 0.000 kB/s
		FTP enabled: YES, TX enabled: YES
		mode: Minimal
		MAVLink version: 1
		transport protocol: serial (/dev/ttyS1 @57600)

	instance #2:
		mavlink chan: #2
		type:           GENERIC LINK OR RADIO
		flow control: OFF
		rates:
		  tx: 0.784 kB/s
		  txerr: 0.000 kB/s
		  tx rate mult: 0.312
		  tx rate max: 1000 B/s
		  rx: 0.000 kB/s
		FTP enabled: YES, TX enabled: YES
		mode: OSD
		MAVLink version: 1
		transport protocol: serial (/dev/ttyS2 @57600)

LorenzMeier, 2019-01-01:
Almost certainly some type of busy loop in the message send-scheduler or alike. @dagar @TSC21 @julianoes as you are altering the system on these fronts: You *cannot* run a normal poll loop against the file handles for all the topics. That proved too inefficient (because of the overhead of `poll()`). So what we need to do is to invert the pattern and instead run at a fixed rate. That is what the implementation is doing, but you can break the architecture unknowingly if you add a "naive" poll implementation.

dagar, 2019-01-01:
I found the problem, a param update slipped into `MavlinkStream::update()` as part of the PR adding an odometry stream. https://github.com/PX4/Firmware/commit/cfd1be584eb3240632adbc6ef803c1ef976d621a#diff-b168380964a16cd9b42995acc531f1e5R61

RE: polling I agree (there's been no fundamental change in that here), although one interesting case I think we should try is having a mavlink VIO mode where the main TX loop polls only the odometry publication rather than sleeping.

TSC21, 2019-01-01:
> polling I agree (there's been no fundamental change in that here), although one interesting case I think we should try is having a mavlink VIO mode where the main TX loop polls only the odometry publication rather than sleeping.

This is the best approach for sure.

commit 8a4b97a8 (2019-01-01): mavlink handle MAV_ODOM_LP parameter in main

 - fixes #11125
commit 55e3f80b (2019-01-01): mavlink handle MAV_ODOM_LP parameter in main

 - fixes #11125
closed 2019-01-01 by TSC21