PX4/PX4-Autopilot #11133: Mavlink_receiver.cpp doesn't include signal_quality for distance_sensor messages

tuloski, 2019-01-03:
distance_sensor uORB message includes a signal_quality field.
In the mavlink_receiver.cpp when some mavlink messages (as DISTANCE_SENSOR #132) should be translated into uORB messages, that field is not filled, causing the message to not be accepted as valid.

It is also not very clear how to generate that field from mavlink messages, since there is not a field in the mavlink message that can be translated into a signal_quality. Just put 100 (perfect signal)? Or -1? Derive a value from covariance?

Example of missing field: https://github.com/PX4/Firmware/blob/73ed9deac507c86892afadee5bca2024b5b8da17/src/modules/mavlink/mavlink_receiver.cpp#L824-L834

julianoes, 2019-07-10:
Thanks for the issue, sorry for the late response. I would vote "Or -1?" given the fact that it's unknown. Are you interested in making a pull request?

tuloski, 2019-07-10:
Yes I can make a pull request.
Probably in a few days.

mcsauder, 2019-10-08:
I just dug into this, there is a mismatch between the `distance_sensor_t` definition and the `mavlink_distance_sensor_t` definition.  Here are the fields missing from the `mavlink_msg_distance_sensor.h` definition of mavlink_distance_sensor_t:
```
ds.h_fov            = dist_sensor.h_fov;
ds.v_fov            = dist_sensor.v_fov;
ds.q                = dist_sensor.q;
ds.signal_quality   = dist_sensor.signal_quality;
```

In order to remedy this we will need to get these fields added to the mavlink message definition.  Here is the place to start: https://github.com/mavlink/rfcs.

I've opened PR #13131 to address this issue and will open an RFC in the mavlink rfc repo as soon as I can to propose a future change.

tuloski, 2019-10-08:
@mcsauder I don't think MAVlink should be modified. It's the autopilot that should adapt to MAVlink, unless those fields are really important for that message type.

TSC21, 2019-10-08:
> @mcsauder I don't think MAVlink should be modified. It's the autopilot that should adapt to MAVlink, unless those fields are really important for that message type.

The message fields are not modified: you can extend the message by adding new fields. I don't see any issue on adding a sensor quality field on the Mavlink message (the other three already exist).

Closing this as it was solved in #13131.

closed 2019-10-08 by TSC21

mcsauder, 2019-10-08:
Thanks @tuloski and @TSC21 !