PX4/PX4-Autopilot #11144: PX4 SITL mission tests intermittent failures

dagar, 2019-01-04:
Examples
 - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-11136/4/pipeline/

dagar, 2019-01-04:
Gazebo plugin crashing? 

[image]

dagar, 2019-01-04:
Different issue, arming failing due to accels.

[image]

dagar, 2019-01-08:
Another instance - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/pr-lndmc-takeoff/1/pipeline

Any thoughts?

dagar, 2019-01-09:
FYI @julianoes

julianoes, 2019-01-10:
I see this ros log output:
```
[rosmaster.master][INFO] 2019-01-10 14:27:18,848: publisherUpdate[/tf] -> http://58c98d63fdec:34145/ []
[rosmaster.master][INFO] 2019-01-10 14:27:18,848: publisherUpdate[/tf] -> http://58c98d63fdec:34145/ []: sec=0.00, exception=<Fault -1: 'publisherUpdate: unknown method name'>
[rosmaster.threadpool][ERROR] 2019-01-10 14:27:18,849: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'>

[rosmaster.master][INFO] 2019-01-10 14:27:18,849: publisherUpdate[/tf_static] -> http://58c98d63fdec:34145/ []
[rosmaster.master][INFO] 2019-01-10 14:27:18,849: publisherUpdate[/tf_static] -> http://58c98d63fdec:34145/ []: sec=0.00, exception=<Fault -1: 'publisherUpdate: unknown method name'>
[rosmaster.threadpool][ERROR] 2019-01-10 14:27:18,850: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'>
```

julianoes, 2019-01-10:
I'm still trying to find out which process crashes. I don't get a ".core" file because it's inside docker, so I'm going to try in native Ubuntu.

dagar, 2019-01-10:
You should still be able to get a core within docker. Is it not being generated at all or a volume issue? 
Start by checking the ulimit within the container (ulimit -c unlimited; ulimit -a).

julianoes, 2019-01-11:
Maybe this wasn't such a good idea for systems where we are slow in the beginning:
https://github.com/PX4/sitl_gazebo/blob/a912715debd3b84b674969120688b2def80e264f/src/gazebo_mavlink_interface.cpp#L1017-L1020

Should be fixed in https://github.com/PX4/sitl_gazebo/pull/265.

PR #11196 (2019-01-11): sitl_gazebo: update submodule
This should fix the CI failures where we intermittently saw SIGABRT.

Fixes #11144.
commit 2e3fa30c (2019-01-12): sitl_gazebo: update to revision that doesn't abort on timeout (#11196)

 * this should fix the CI failures where we intermittently saw SIGABRT. (fixes #11144)
closed 2019-01-11 by julianoes