PX4/PX4-Autopilot #11146: Unable to build v1.8.2 from source

Seeelefant, 2019-01-05:
**Describe the bug**
```
make px4_fmu-v3
Makefile:378: *** "px4_fmu-v3 cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Schluss.
```

**To Reproduce**
Steps to reproduce the behavior:
Following the instructions from [Building PX4 Software](https://dev.px4.io/en/setup/building_px4.html) I get the error message above. Please advice.

```
$  git clone https://github.com/PX4/Firmware.git
Klone nach 'Firmware' ...
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (53/53), done.
remote: Total 272154 (delta 35), reused 27 (delta 16), pack-reused 272085
Empfange Objekte: 100% (272154/272154), 103.62 MiB | 4.75 MiB/s, Fertig.
Löse Unterschiede auf: 100% (204412/204412), Fertig.
Checke Dateien aus: 100% (2522/2522), Fertig.

Wolfgang@PC-Wolfgang ~/Firmware
$ git checkout v1.8.2
Checke Dateien aus: 100% (2476/2476), Fertig.
Hinweis: Checke 'v1.8.2' aus.

Sie befinden sich im Zustand eines 'losgelösten HEAD'. Sie können sich
umschauen, experimentelle Änderungen vornehmen und diese committen, und
Sie können alle möglichen Commits, die Sie in diesem Zustand machen,
ohne Auswirkungen auf irgendeinen Branch verwerfen, indem Sie einen
weiteren Checkout durchführen.

Wenn Sie einen neuen Branch erstellen möchten, um Ihre erstellten Commits
zu behalten, können Sie das (jetzt oder später) durch einen weiteren Checkout
mit der Option -b tun. Beispiel:

  git checkout -b <neuer-Branchname>

HEAD ist jetzt bei f13bbacd52 mc_att_control: copy sensor_correction topic once initially

$ make px4_fmu-v3
Makefile:378: *** "px4_fmu-v3 cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Schluss.
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

TSC21, 2019-01-05:
Use `make px4fmu-v3_default`. The new `make` targets do not work on 1.8.2 stable. They will be on 1.9.0 stable release.

closed 2019-01-05 by TSC21

Seeelefant, 2019-01-05:
Thanks, at least the make px4fmu-v3_default was working. 

- Trying to upload the generated file via qgroundcontrol reproducible leads to 
[image]

- Trying to upload the "last stable version 1.8.2" via qgroundcontrol always flashes (without any further problems) version 1.8.1

- Your master brach seems to bee quite unstable, whereas the recommended approach is to introduce a development branch and keep the master branch stable according to [A successful Git branching model](https://nvie.com/posts/a-successful-git-branching-model/)

Please advice
--timonito

TSC21, 2019-01-05:
https://dev.px4.io/en/setup/building_px4.html#uploading-firmware-flashing-the-board

Seeelefant, 2019-01-05:
I´ve read that, but unfortunately 

> You can flash the custom firmware on Windows using QGroundControl or Mission Planner (it is not possible to directly flash the firmware from within the bash shell using the upload command).

according [to](https://dev.px4.io/en/setup/dev_env_windows_bash_on_win.html), so I tried it with QGroundcontrol. 

Please advice
--Seeelefant

Seeelefant, 2019-01-06:
Just to add something here: I tried in Ubuntu VM, after running trough all the steps above I got 

```
wolfgang@wolfgang-VirtualBox:~/src/Firmware$ git checkout v1.8.2
Note: checking out 'v1.8.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at f13bbac... mc_att_control: copy sensor_correction topic once initially
wolfgang@wolfgang-VirtualBox:~/src/Firmware$ make px4fmu-v3_default
-- PX4 VERSION: v1.8.2
-- CONFIG: nuttx_px4fmu-v3_default
-- Build Type: MinSizeRel
CMake Error at cmake/toolchains/Toolchain-arm-none-eabi.cmake:62 (message):
  could not find genromfs
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CMakeDetermineSystem.cmake:98 (include)
  CMakeLists.txt:176 (project)

CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make[1]: *** /home/wolfgang/src/Firmware/build/nuttx_px4fmu-v3_default: No such file or directory.  Stop.
Makefile:154: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 2
wolfgang@wolfgang-VirtualBox:~/src/Firmware$ CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage

```

Please advice
--Seeelefant

hamishwillee, 2019-01-06:
@Seeelefant Did you run the ubuntu_sim_nuttx.sh setup script in your VM as described here: https://dev.px4.io/v1.8.2/en/setup/dev_env_linux.html

NOTE this is the v1.8.2 devguide. Instructions should work if followed correctly (this is my dev environment)

Seeelefant, 2019-01-07:
Thanks Hamish, now I got the build working. For me it is a little confusing that for PX4 the master branch is the unstable one.
What remains is the problem with the firmware upload via Qgroundcontrol:
[image]
Can you help in this direction?

hamishwillee, 2019-01-08:
No idea sorry. That should be a separate issue.

> For me it is a little confusing that for PX4 the master branch is the unstable one.

Just the way it is. I've worked in companies that use this approach, and others that have a develop branch that goes into a master periodically following certain tests. The approach PX4 uses is more lightweight, and I think "getting better" as we improve our continuous integration. Not perfect though.

Seeelefant, 2019-01-08:
Thanks Hamish, just as a little remark [ A successful Git branching model](https://nvie.com/posts/a-successful-git-branching-model/), seems more systematic and reliable to me ;-)
Best regards
--Seeelefant

Seeelefant, 2019-01-20:
Solved the problem with firmware upload, seems to be a timing problem. The connection to the FMU was running through an USB hub, after connecting directly to the PC it worked.