PX4/PX4-Autopilot #11150: Unable to biuld px4 on beaglebone blue nativ

7st4r, 2019-01-05:
Does anyone have an insight in to this problem 

[ 98%] Built target FlightTasks
Scanning dependencies of target modules__mc_pos_control
[ 98%] Building CXX object src/modules/mc_pos_control/CMakeFiles/modules__mc_pos_control.dir/mc_pos_control_main.cpp.o
[ 98%] Building CXX object src/modules/mc_pos_control/CMakeFiles/modules__mc_pos_control.dir/PositionControl.cpp.o
[100%] Building CXX object src/modules/mc_pos_control/CMakeFiles/modules__mc_pos_control.dir/Utility/ControlMath.cpp.o
[100%] Linking CXX static library libmodules__mc_pos_control.a
[100%] Built target modules__mc_pos_control
Scanning dependencies of target px4
[100%] Building CXX object platforms/posix/CMakeFiles/px4.dir/src/main.cpp.o
[100%] Building CXX object platforms/posix/CMakeFiles/px4.dir/apps.cpp.o
[100%] Linking CXX executable ../../bin/px4
../../src/lib/DriverFramework/framework/src/libdf_driver_framework.a(I2CDevObj.cpp.o): In function `DriverFramework::I2CDevObj::_readReg(unsigned char, unsigned char*, unsigned int)':
/var/lib/cloud9/Firmware/src/lib/DriverFramework/framework/src/I2CDevObj.cpp:174: undefined reference to `rc_i2c_read_data'
collect2: error: ld returned 1 exit status
platforms/posix/CMakeFiles/px4.dir/build.make:262: recipe for target 'bin/px4' failed
make[3]: *** [bin/px4] Error 1
CMakeFiles/Makefile2:10494: recipe for target 'platforms/posix/CMakeFiles/px4.dir/all' failed
make[2]: *** [platforms/posix/CMakeFiles/px4.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:171: recipe for target 'beaglebone_blue_native' failed
make: *** [beaglebone_blue_native] Error 2

dagar, 2019-01-08:
There's been a lot of refactoring of general board support. This board was never added to the build system and the test team doesn't have one, so I'm not surprised it's currently broken.

dagar, 2019-01-08:
@UAV-Pilot could you help get this working? I'd then like to get it added to Jenkins so we can keep it working. Example - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile/detail/master/493/pipeline

7st4r, 2019-01-09:
@UAV-Pilot I would very much appriciat if you have any insight on getting this work.

UAV-Pilot, 2019-01-09:
@7st4r If I remember correctly, I tested native build before documenting the process here: https://docs.px4.io/en/flight_controller/beaglebone_blue.html. Did you set up LIBROBOTCONTROL_INSTALL_DIR, etc as described in "Setup Robot Control Library" section?

7st4r, 2019-01-09:
@UAV-Pilot I followed that guid and it would not finish the make process. Have you tested it recently? There are several others who have had similar issue. Any way i would apprciat for any solution as i have been stuck on it for some time now.

UAV-Pilot, 2019-01-11:
@7st4r  It looks like handling for a macro is missing in the native build for the Robot Control Library. Try the following:
1. insert $(EXT_CFLAGS) into CFLAGS in librobotcontrol/library/Makefile, e.g.,
CFLAGS          := $(EXT_CFLAGS) -g -fPIC -I $(INCLUDEDIR)

2. include the macro definition in the make call, e.g.,
make EXT_CFLAGS=-DRC_AUTOPILOT_EXT ......

UAV-Pilot, 2019-02-17:
Documented the fix in this pull request: https://github.com/PX4/px4_user_guide/pull/438

UAV-Pilot, 2019-02-17:
The fix is in this pull request: https://github.com/StrawsonDesign/librobotcontrol/pull/148

closed 2019-02-17 by LorenzMeier