PX4/PX4-Autopilot #11151: issues of jmavsim

roseyangyu, 2019-01-06:
I have simulation used jmavsim.But I am failed.

`ERROR [param] importing from 'rootfs/eeprom/parameters' failed (-1)
Command 'param' failed, returned 1
INFO  [dataman] Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11405132 bytes
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1376264
  CAL_ACC0_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_XSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC0_YOFF: curr: 0.0000 -> new: -0.0100
  CAL_ACC0_YSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC0_ZOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_ZSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC1_ID: curr: 0 -> new: 1310728
  CAL_ACC1_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_GYRO0_ID: curr: 0 -> new: 2293768
  CAL_GYRO0_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_MAG0_ID: curr: 0 -> new: 196616
  CAL_MAG0_XOFF: curr: 0.0000 -> new: 0.0100
  COM_DISARM_LAND: curr: 0 -> new: 3
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_MAG_TYPE: curr: 0 -> new: 1
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.0200 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SYS_AUTOSTART: curr: 0 -> new: 4010
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
Buildfile: /home/airwing/src/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x6cb20e63, isOwner false, <11be88f1, 3b08f7e6>[count 0, qsz 0, owner <NULL>]]]
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
	... 2 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped 
	at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2047)
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1917)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1866)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1835)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:79)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:229)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:215)
	at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2272)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1020)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1049)
	at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:802)
	at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:132)
	at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
	at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:837)
	at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:274)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:154)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
	... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7f2d380010d0, refCount 1, unCloseable false]
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x6cb20e63, isOwner false, <11be88f1, 3b08f7e6>[count 0, qsz 0, owner <NULL>]]]
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
	... 2 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped 
	at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2047)
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1917)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1866)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1835)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:79)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:229)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:215)
	at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2272)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1020)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1049)
	at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:802)
	at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:132)
	at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
	at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:837)
	at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:274)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:154)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
	... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7ff92c001390, refCount 1, unCloseable false]

`
FW version 1.7.0
I need help!

roseyangyu, 2019-01-07:
I use ubuntu16.04.I run this scripts of [Development Environment on Ubuntu LTS / Debian Linux](http://dev.px4.io/en/setup/dev_env_linux_ubuntu.html).ubuntu_sim_common_deps.sh,ubuntu_sim.sh,ubuntu_sim_nuttx.sh and ubuntu_sim_ros_gazebo.sh.

hamishwillee, 2019-01-07:
On a VM or a real machine?

roseyangyu, 2019-01-07:
a real machine

roseyangyu, 2019-01-07:
I have tried v1.8.0, is same as that issues.I have re-install system-ubuntu 16.04,I have only run ubuntu_sim_nuttx.sh.But Issues still exist. FW 1.8.0 and 1.7.0.

hamishwillee, 2019-01-07:
Thanks. We're now beyond what I know. Lets see what @bkueng et al can come up with to help.

bkueng, 2019-01-07:
Duplicate of https://github.com/PX4/Firmware/issues/9283.

Looks like an issue with the graphics driver, and you can try installing a different one. Which driver an GPU do you use? Does it have 3D acceleration enabled?

roseyangyu, 2019-01-07:
`WARNING: you should run this program as super-user.
  *-display               
       description: VGA compatible controller
       product: Carrizo
       vendor: Advanced Micro Devices, Inc. [AMD/ATI]
       physical id: 1
       bus info: pci@0000:00:01.0
       version: c9
       width: 64 bits
       clock: 33MHz
       capabilities: vga_controller bus_master cap_list rom
       configuration: driver=amdgpu latency=0
       resources: irq:39 memory:e0000000-efffffff memory:d0800000-d0ffffff ioport:4000(size=256) memory:d1700000-d173ffff memory:c0000-dffff
WARNING: output may be incomplete or inaccurate, you should run this program as super-user.
`

roseyangyu, 2019-01-07:
I have followed #9283 .But I can't solve that

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
@roseyangyu did you find a solution in the meantime?

roseyangyu, 2019-07-25:
@julianoes I can't.But I find that if I have simlation on same GPU in the computer,it's still there.So I think that  problem has to do with GPU(AMD).Nvidia never show any problems

ChoudharyRamesh, 2022-07-16:
Solved [here](https://github.com/PX4/PX4-Autopilot/issues/18743#issuecomment-1186202454)