PX4/PX4-Autopilot #11156: Gazebo does work when simulating multiple UAVs

SamNew1, 2019-01-06:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

aomezzine, 2019-01-16:
Hi,
I have the same problem when launching  w multi_uav_mavros_sitl.
Note that it works fine with an older version of the Firmware (downloaded on December).
Thank you for your help.

billiepratama, 2019-01-18:
@SamNew1 Hello, i got the exactly same problem with you. Gazebo got stuck and just preparing the world. Then, i tried to clean by "make clean" in the directory Firmware directory. I tried to roslaunch px4 multi_uav_mavros_sitl.launch and it worked. But, now the problem is mavros topics don't publish any values. I hope that anyone can help this issue.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
Did you have any success fixing this? As far as I know this should be working.

billiepratama, 2019-07-10:
What is your problem? Maybe i can help you.
In my case, i tried to upgrade my ubuntu version to 18.04 and install ROS
Melodic. After that, everything works fine.