PX4/PX4-Autopilot #11158: mappingh thrust force into thrust setpoint

robin-ecn, 2019-01-07:
Hi guys,

I am controling a drone (`px4 posix_sitl.launch`) by ROS in gazbo simulation. The communication is through the mavros.

From the controller, I get the thrust force needed and I poublish to the topic ~setpoint_raw/attitude (http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_raw).

However, the the thrust force I got is in Newton but the robot in gazbo only accept a normalized thrust vale (0-1). Since it is a simulation in gazbo, I cannot do a dynamic identification...

Is there anyway for me to get the mapping the thrust force into the normalised thrust vale?

Thanks

Robin

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
@bresch this question is for you.

sourishg, 2019-11-05:
@robin-ecn I'm also facing the same issue as you were. Could you figure out how to find the mapping?