PX4/PX4-Autopilot #11162: mpu9250 not starting on px4_fmu-v4 (and likely others)

dagar, 2019-01-07:
This is likely more fallout from expanding the existing mpu9250 driver to support the icm20948 (https://github.com/PX4/Firmware/pull/10116) and followup fixes (https://github.com/PX4/Firmware/pull/11020)

Noticed by Jenkins (http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/pr-update_src%2Flib%2Fecl/70/pipeline/79).

[image]

FYI @dakejahl @DanielePettenuzzo @flochir

dakejahl, 2019-01-07:
Do these buses have the mpu9250/6500/20948 on them? I noticed the driver was now trying to start on every bus and just failing if device not present.

dagar, 2019-01-07:
That's an mRo pixracer on the hardware test rack with only an mRo GPS connected (https://store.mrobotics.io/mRo-GPS-u-Blox-Neo-M8N-HMC5983-Compass-p/mro-gps003-mr.htm), dronecode debug (SWD + serial console), and USB power.

dakejahl, 2019-01-07:
Does that variant have the mpu9250 on i2c? That screenshot is of i2c start failing

dakejahl, 2019-01-07:
Looks like the mpu9250 is started without a `bus_id` opt arg and thus it attempts to start on all busses.
https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v4/init/rc.board#L84-L89

dakejahl, 2019-01-07:
[image]
And bus 3 corresponds to `MPU9250_BUS_SPI_INTERNAL`

dakejahl, 2019-01-07:
This is with a pixracer, not sure the version (not printed on the silk screen) but it's one of those little red boards

flochir, 2019-01-08:
Omitting the bus on the driver call shouldn't cause a fail though, just a chance of starting an unintended device with maybe wrong parameters.

flochir, 2019-01-09:
@dakejahl Ok, I see it now. I think that happens now because the probe() function of mpu9250_i2c.cpp was disabled in #11020. It gets called implicitly by interface->init() in line 232 of main.cpp. This way, the driver itself only got instanciated when the device was already detected, and a fail of init() actually meant a fail. Now that can also be caused by the device not being present.

flochir, 2019-01-09:
Added a new whoami-based version of device probe in I2C interface probe to PR: #11173
@dagar @dakejahl Can you try if this fixes the issue?

dagar, 2019-01-09:
Fixed by https://github.com/PX4/Firmware/pull/11173

closed 2019-01-09 by dagar