PX4/PX4-Autopilot #11175: v1.9.0 Release notes

mrpollo, 2019-01-09:
WIP

---

# Change Log

## [v1.9.0](https://github.com/PX4/Firmware/tree/v1.9.0) (2019-01-08)
[Full Changelog](https://github.com/PX4/Firmware/compare/v1.8.2...v1.9.0)

**Implemented enhancements:**

- add more relevant messages to the RTPS stream [\#11120](https://github.com/PX4/Firmware/pull/11120) ([TSC21](https://github.com/TSC21))
- Feature: VIO: add ODOMETRY stream [\#11084](https://github.com/PX4/Firmware/pull/11084) ([TSC21](https://github.com/TSC21))
- increase pmw3901 publish rate and max\_ground\_distance [\#11066](https://github.com/PX4/Firmware/pull/11066) ([AnnaDaiZH](https://github.com/AnnaDaiZH))
- SBUS parsing: add time-based hardening on IO [\#11040](https://github.com/PX4/Firmware/pull/11040) ([bkueng](https://github.com/bkueng))
- Make SITL tests faster than real-time [\#10648](https://github.com/PX4/Firmware/pull/10648) ([julianoes](https://github.com/julianoes))
- Orbit: Send telemetry for the UI [\#10989](https://github.com/PX4/Firmware/pull/10989) ([MaEtUgR](https://github.com/MaEtUgR))
- Update MacOSX install script [\#10966](https://github.com/PX4/Firmware/pull/10966) ([TSC21](https://github.com/TSC21))
- MC: Multirotor mixer rewrite & Airmode for Yaw [\#10962](https://github.com/PX4/Firmware/pull/10962) ([bkueng](https://github.com/bkueng))
- allow boards to supply an rc.board init [\#10960](https://github.com/PX4/Firmware/pull/10960) ([dagar](https://github.com/dagar))
- update micro-CDR version and msg templates to fit the up to date version [\#10954](https://github.com/PX4/Firmware/pull/10954) ([TSC21](https://github.com/TSC21))
- Check NuttX configuration for DMA and load monitoring [\#10952](https://github.com/PX4/Firmware/pull/10952) ([christianrauch](https://github.com/christianrauch))
- fast-RTPS: IDL generation: add constants fields for IDL's [\#10949](https://github.com/PX4/Firmware/pull/10949) ([TSC21](https://github.com/TSC21))
- \[WIP\] vscode configuration [\#10911](https://github.com/PX4/Firmware/pull/10911) ([dagar](https://github.com/dagar))
- Enable Orbit Flight Mode [\#10907](https://github.com/PX4/Firmware/pull/10907) ([MaEtUgR](https://github.com/MaEtUgR))
- Windows CI: run `make tests` [\#10890](https://github.com/PX4/Firmware/pull/10890) ([MaEtUgR](https://github.com/MaEtUgR))
- uORB replace ORBMap with linked list [\#10887](https://github.com/PX4/Firmware/pull/10887) ([dagar](https://github.com/dagar))
- boards organization [\#10875](https://github.com/PX4/Firmware/pull/10875) ([dagar](https://github.com/dagar))
- Appveyor: Switch build cache to the entire PX4 folder [\#10854](https://github.com/PX4/Firmware/pull/10854) ([MaEtUgR](https://github.com/MaEtUgR))
- Landing gear improvement [\#10842](https://github.com/PX4/Firmware/pull/10842) ([MaEtUgR](https://github.com/MaEtUgR))
- Use unix sockets instead of pipes for posix daemon. [\#10766](https://github.com/PX4/Firmware/pull/10766) ([m-ou-se](https://github.com/m-ou-se))
- Tracking anti-reset windup for multicopter velocity controller [\#10708](https://github.com/PX4/Firmware/pull/10708) ([bresch](https://github.com/bresch))
- ICM20948 integration into MPU9250 driver [\#10116](https://github.com/PX4/Firmware/pull/10116) ([flochir](https://github.com/flochir))

**Fixed bugs:**

- HITL regression with jMAVSim [\#11165](https://github.com/PX4/Firmware/issues/11165)
- mavlink 0 \(USB\) abnormal cpu usage [\#11125](https://github.com/PX4/Firmware/issues/11125)
- Commander spams the nsh console on SITL [\#11118](https://github.com/PX4/Firmware/issues/11118)
- Snapdragon build fails at runtime \(related to time functions\) [\#11110](https://github.com/PX4/Firmware/issues/11110)
- PR \#10116 breaks arbitrary bus/device compatibility [\#11019](https://github.com/PX4/Firmware/issues/11019)
- Building as ROS Catkin Package Install Broken [\#11006](https://github.com/PX4/Firmware/issues/11006)
- RC Can't generate PWM output and rotor not spin in PX4 pixhawk1 [\#10997](https://github.com/PX4/Firmware/issues/10997)
- pixhack v5 RC is not detected on master  [\#9989](https://github.com/PX4/Firmware/issues/9989)
- Restore dsp\_offset and remove unused qurt code. Fixes \#11134 [\#11139](https://github.com/PX4/Firmware/pull/11139) ([karliss](https://github.com/karliss))
- rgbled\_ncp5623c set main stack 1500 [\#11135](https://github.com/PX4/Firmware/pull/11135) ([dagar](https://github.com/dagar))
- mavlink fix performance regression \(\#11125\), add interval perf counter, and cleanup getters [\#11126](https://github.com/PX4/Firmware/pull/11126) ([dagar](https://github.com/dagar))
- platforms: fix linking for Snapdragon [\#11115](https://github.com/PX4/Firmware/pull/11115) ([julianoes](https://github.com/julianoes))
- Fix pwm sim publishing when disarmed [\#11112](https://github.com/PX4/Firmware/pull/11112) ([MaEtUgR](https://github.com/MaEtUgR))
- commander set home position timestamp properly [\#11109](https://github.com/PX4/Firmware/pull/11109) ([dagar](https://github.com/dagar))
- Bug in checking if landing point has been passed [\#11083](https://github.com/PX4/Firmware/pull/11083) ([almaaro](https://github.com/almaaro))
- Small practical Orbit fixes [\#11069](https://github.com/PX4/Firmware/pull/11069) ([MaEtUgR](https://github.com/MaEtUgR))
- investigate px4fmu-v5 cpu spikes [\#9852](https://github.com/PX4/Firmware/pull/9852) ([dagar](https://github.com/dagar))
- fmu-v5: ERROR \[pwm\] can't open /dev/pwm\_output0 [\#10981](https://github.com/PX4/Firmware/issues/10981)
- Issue with Master [\#10980](https://github.com/PX4/Firmware/issues/10980)
- BUG: QGC does not show most parameters [\#10972](https://github.com/PX4/Firmware/issues/10972)
- Master branch firmware \(fmu v3\) cannot drive all sensors [\#10961](https://github.com/PX4/Firmware/issues/10961)
- tone\_alarm is not compiled anymore [\#10925](https://github.com/PX4/Firmware/issues/10925)
- Bad param meta data: VT\_AIRSPD\_RULE is UINT but min is set with float value [\#10896](https://github.com/PX4/Firmware/issues/10896)
- macOS SITL broken after daemon change [\#10839](https://github.com/PX4/Firmware/issues/10839)
- Verify functionality of sensor disable parameters [\#10818](https://github.com/PX4/Firmware/issues/10818)
- Iridium driver crashes if device not connected [\#10092](https://github.com/PX4/Firmware/issues/10092)
- Pixhawk 4 Mini \(fmu-v5\) minor issues [\#10034](https://github.com/PX4/Firmware/issues/10034)
- Fix snapdragon build [\#11000](https://github.com/PX4/Firmware/pull/11000) ([karliss](https://github.com/karliss))
-  ROMFS multicopter airframes don't call other airframes [\#10983](https://github.com/PX4/Firmware/pull/10983) ([dagar](https://github.com/dagar))
- Use configurable vertical smoothing also in altitude mode [\#10928](https://github.com/PX4/Firmware/pull/10928) ([MaEtUgR](https://github.com/MaEtUgR))
- remove unused VT\_AIRSPD\_RULE [\#10897](https://github.com/PX4/Firmware/pull/10897) ([dagar](https://github.com/dagar))
- Makefile handle spaces in path [\#10895](https://github.com/PX4/Firmware/pull/10895) ([dagar](https://github.com/dagar))
- \[bug fix\] Covariance check on INAV and AEQ [\#10872](https://github.com/PX4/Firmware/pull/10872) ([TSC21](https://github.com/TSC21))
- EKF: Magnetic declination use fixes [\#10848](https://github.com/PX4/Firmware/pull/10848) ([priseborough](https://github.com/priseborough))
[... 82 lines cut ...]
- Update submodule sitl\_gazebo to latest Wed Nov 21 00:37:52 UTC 2018 [\#10886](https://github.com/PX4/Firmware/pull/10886) ([dagar](https://github.com/dagar))
- Update submodule mavlink v2.0 to latest Wed Nov 21 00:38:04 UTC 2018 [\#10885](https://github.com/PX4/Firmware/pull/10885) ([dagar](https://github.com/dagar))
- Update submodule matrix to latest Wed Nov 21 00:38:27 UTC 2018 [\#10884](https://github.com/PX4/Firmware/pull/10884) ([dagar](https://github.com/dagar))
- Fix for initializing Landing Target Estimator in SITL [\#10882](https://github.com/PX4/Firmware/pull/10882) ([ebabyak](https://github.com/ebabyak))
- Small px\_romfs\_pruner.py fixes [\#10880](https://github.com/PX4/Firmware/pull/10880) ([julianoes](https://github.com/julianoes))
- Add UTM\_GLOBAL\_POSITION MAVLink stream [\#10873](https://github.com/PX4/Firmware/pull/10873) ([eyeam3](https://github.com/eyeam3))
- Add dynamic modules on Posix. [\#10870](https://github.com/PX4/Firmware/pull/10870) ([m-ou-se](https://github.com/m-ou-se))
[... cut ...]
- mavlink: fix yawing to North for LOITER\_TIME [\#10828](https://github.com/PX4/Firmware/pull/10828) ([julianoes](https://github.com/julianoes))
- VTOL rate control improvements [\#10819](https://github.com/PX4/Firmware/pull/10819) ([RomanBapst](https://github.com/RomanBapst))
- Add a collision avoidance library which uses range data [\#10785](https://github.com/PX4/Firmware/pull/10785) ([baumanta](https://github.com/baumanta))

\* *This Change Log was automatically generated by [github_changelog_generator](https://github.com/skywinder/Github-Changelog-Generator)*

mrpollo, 2019-01-09:
@dagar this is the output of the changelog generator tool, let me know if there's anything missing

hamishwillee, 2019-01-09:
I'm not sure about how useful that is and what is enhancements vs PRs. The fact that it misses that we implemented Collision Prevention and Obstacle avoidance in missions is a worry.

hamishwillee, 2019-01-09:
@mrpollo It might not have missed them, just put them in wrong place - and the PR titles don't really capture the significance of particular features. #10785 is collision prevention. Not sure about obstacle avoidance.

Antiheavy, 2019-01-09:
this one seems significant: https://github.com/PX4/Firmware/pull/10738

TSC21, 2019-01-10:
@dakejahl ECL is platform independent.

mrpollo, 2019-01-11:
@dakejahl @TSC21 If the ECL submodule was updated it should be reflected here in some way and we can link to the ECL project for details on the update, same for every other submodule, thanks for bringing this up.

@hamishwillee I totally see where you are coming from here, I'm with you on this one `changelog != release notes`.

E666GT, 2019-01-13:
Great Work!

mrpollo, 2019-04-10:
Just pushed an update!

jinger26, 2019-04-17:
@dagar , change log is updated now, we need to distill this into high level release note.

hamishwillee, 2019-04-30:
@mrpollo @MaEtUgR Is orbit mode going in? https://github.com/PX4/Firmware/pull/10907

1Gump, 2019-04-30:
I don't know if this is valid for this this log, but a big reason I need this update is for the **lidarlite v3hp functionality** - it didn't work in 1.8.2

MaEtUgR, 2019-05-03:
@hamishwillee Orbit is in and QGC also supports the command since last release. Good catch, I don't know where it got lost.

mrpollo, 2019-05-03:
That's a great catch @hamishwillee 

Everyone, please help us review the changelog for any missing features, we should be as accurate as possible.

hamishwillee, 2019-05-27:
@mrpollo You are planning on creating a high level overview part to these notes right - in line with https://discuss.px4.io/t/what-to-expect-from-1-9-release/11117/3 ?

closed 2019-08-23 by julianoes