PX4/PX4-Autopilot #11182: Problem installing Gazebo on Ubuntu "FAILED: CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o  /usr/bin/c++"

Tanop74, 2019-01-09:
I use Linux Ubuntu Ubuntu 18.04.1 LTS and QGC with PX4 and it works well.

I wanted to try so do simulation on Gazebo.

When I follow the installation process of Gazebo it works well until I get the following problems:

[12/19] Building CXX object CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o
FAILED: CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_imu_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.1 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -I/home/tanop/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/tanop/src/Firmware/mavlink/include -I/usr/include/Paging -isystem /usr/include/opencv -I/home/tanop/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o -MF CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o.d -o CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o -c /home/tanop/src/Firmware/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp
virtual memory exhausted: Cannot allocate memory
[13/19] Building CXX object CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o
FAILED: CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_lidar_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.1 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -I/home/tanop/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/tanop/src/Firmware/mavlink/include -I/usr/include/Paging -isystem /usr/include/opencv -I/home/tanop/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o -MF CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o.d -o CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o -c /home/tanop/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp
virtual memory exhausted: Cannot allocate memory
[14/19] Building CXX object CMakeFiles/gazebo_irlock_plugin.dir/src/gazebo_irlock_plugin.cpp.o
[15/19] Building CXX object CMakeFiles/gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:182: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Can you please help me?
Thanks

TSC21, 2019-01-09:
```
virtual memory exhausted: Cannot allocate memory
```

Where exactly are you actually trying to build this?

TSC21, 2019-01-09:
Try increasing your swap space first before building the repo.

TSC21, 2019-01-09:
https://help.ubuntu.com/community/SwapFaq

closed 2019-01-09 by TSC21