PX4/PX4-Autopilot #11183: Mission Protocol does not properly handle cancelation

hamishwillee, 2019-01-09:
The only way (currently) supported by MAVLink to cancel a mission is to NACK a message (MISSION_ACK) with an error. Currently if there is an error the system will report a [critical error](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L606). 

The required change is that `MAV_MISSION_OPERATION_CANCELLED` ack is received PX4 should not emit a critical error (it might emit some other notification, but this case is "OK", not an error).

FYI The MAV_MISSION_OPERATION_CANCELLED was added by https://github.com/mavlink/mavlink/pull/1044 . Docs will be added to MAVLink spec by https://github.com/mavlink/mavlink-devguide/pull/157

@julianoes  FYI

hamishwillee, 2019-10-01:
@julianoes I have pinned. We probably do need to fix this though.

julianoes, 2026-08-31:
This bot does not care about pinned, does it.

dakejahl, 2026-11-01:
**Claude on behalf of @dakejahl**

Fixed on main by #14139: a MISSION_ACK with MAV_MISSION_OPERATION_CANCELLED ends an upload or download quietly and returns the mission manager to idle.

closed 2026-11-01 by dakejahl