PX4/PX4-Autopilot #11198: Noise (?) on SUMD channels

hangmann, 2019-01-11:
**Describe the bug**

Everytime I connect my SUMD RC receiver to the SUMD interface of the PX4 (with firmware version 1.9), the corresponding channels are read in with huge noise around the actual send values. This noise can be observed in the screenshot below
chan1_raw displays the raw input on the SUMD channel 1. As you can see there are frequently spikes (of partly different heights) in the actually constant sent value of 1500.

For further debug I inspected the decoded SUMD signal, that is published to its subscribers (such as mavlink debug). All decoded SUMD channels (which can be seen in the screenshot as M1:rc_sumd_i) are exactly as they are meant to be - a constant value of 1500.

Long story short: subscribers of the RCInput component receive a faulty, noisy value, despite the channel is received correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect SUMD RC receiver to pixhawk
2. Power on Pixhawk
3. Observe noisy Channels on Channel monitor in QGroundControl

**Expected behavior**
I expected that all subscribers of RCInput should receive the correctly decoded SUMD signal.

**Log Files and Screenshots**
[image]

dagar, 2019-01-11:
@hangmann if you could capture a corresponding log file (.ulg) it might help.

@bkueng or @davids5 - any experience with SUMD?

davids5, 2019-01-11:
@dagar - No - but If need be I can get the equipment to validate. 

@hangmann - what FMU HW is this on?  What is the part numbers for the RC RX and TX used?

bkueng, 2019-01-14:
@dagar no.

@hangmann has this been working before for you?

hangmann, 2019-01-15:
thanks for your responses.

@dagar i will supply it later.
@davids5s this is on FMUv5. I am using Graupner MZ-12 with Falcon 12 receiver
@bkueng this never worked for me

after a short time of this noisy signals, I always experience a crash of the entire SUMD input.

hangmann, 2019-01-16:
Please find the generated ulog file in the attachments. Note, that I have not logged any crash in this file. I will hand it in later. The current log only shows the noise on the rc input.
[px4_sumd_bug.zip](https://github.com/PX4/Firmware/files/2763900/px4_sumd_bug.zip)

hangmann, 2019-02-07:
Could anyone of you reconstruct the issue?
Is there anything you want me to clarify?

julianoes, 2019-07-10:
There are not several reports of RC issues with v1.9. @davids5 do you see a common theme?

davids5, 2019-07-10:
On FMUv5 SUMD is processed by the PX4IO, there was a restructuring in the code to the RC lib. There was also a concern about parity errors not propagating from Nuttx to the app.  We should also check, that the PPM interrupts (on FMU and PX4IO) are off when the scanner has selected a serial protocol.

I would need to get a Graupner MZ-12 and a Falcon 12 receiver to test this.

julianoes, 2019-10-09:
@hangmann which receiver are you using and how is it connected to what Pixhawk? This would help someone to reproduce the problem.

julianoes, 2020-02-12:
Closing until we get an answer.

closed 2020-02-12 by julianoes