PX4/PX4-Autopilot #11203: HITL jmavsim fail

E666GT, 2019-01-13:
**Describe the bug**
I follow http://dev.px4.io/en/simulation/hitl.html
but meet error.
It can obviously get RC signal, I can easily see it arm and disarm in the qground. I can also see it
'MSG: Landing detected'
and
'MSG: Takeoff detected',when I control my RC controller.
but virtual copter moves nothing in the Jmavsim simulator. Virtual copter keeps staying still.

I tested it indoors, is it a problem about GPS thing? If is, why there is also no reponse in Z direction.

**Expected behavior**
Virtual copter moves as I control with RC Controller.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:

Here's the flight log.
[test log 5](https://logs.px4.io/plot_app?log=1ec6070f-2028-4292-b4c1-61336bb4e2c6)
[test log 4](https://logs.px4.io/plot_app?log=4cbc7ec5-7625-40ae-a3f2-b80e6640ea90)
[test log 3](https://logs.px4.io/plot_app?log=2bf8dd8e-1deb-427d-972e-fa1bf658085b)
[test log 2](https://logs.px4.io/plot_app?log=39d6a810-b31b-4735-9bf1-f9e712f2d945)
[test log 1](https://logs.px4.io/plot_app?log=ad447bfa-4857-431a-b89f-f9fe4ca15065)

screenshots
[image]

**Additional context**
Thanks for your team's work

julianoes, 2019-01-17:
What target have you flashed on your hardware? Could it be `px4_fmu-v2_default`?

E666GT, 2019-01-17:
> What target have you flashed on your hardware? Could it be `px4_fmu-v2_default`?

Thank you for focusing on my problem. 

I flashed `px4_fmu-v2_default` .  Because my hardware is Pixhawker.(pixhawk1)

julianoes, 2019-01-17:
Ok, so the problem is that for `v2` there is not enough flash space. What you have to do is to enable `pwm_out_sim` here:
https://github.com/PX4/Firmware/blob/ef54bff4ed45755b4762ef58db21fa37d612d44e/boards/px4/fmu-v2/default.cmake#L50

Then it will overflow of space, so you need to deactivate some other modules such as fixedwing (`fw_`) or vtol (`vtol_`).

julianoes, 2019-01-17:
Also, you can check if you really do have the silicon errata which limits you to 1MB of flash instead of 2MB by doing `ver all` in `nsh>`.

More info here:
https://docs.px4.io/en/flight_controller/silicon_errata.html#fmuv2--pixhawk-silicon-errata
https://docs.px4.io/en/advanced_config/bootloader_update.html

E666GT, 2019-01-17:
Thanks
I follow your advise, unfortunately it doesn't work.
It still have somehow problem.
`CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR`
[image]

PS.
1. I try to update bootloader, it still be v2. I guess it can't be updated to v3.
2. My hardware has 2mb flash, I use
`make px4_fmu-v3_default force-upload ` and
`make px4_fmu-v2_default force-upload ` (for overflowed v2).
But both failed in `CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR`

julianoes, 2019-01-17:
You have this problem #11179.

E666GT, 2019-01-17:
Amazing, I follow your advice: 
`1.Enable pwm_out_sim. 2.And I pull the latest PX4 Firmware as told in #11179 #11104 `
The virtual copter successful flied!

BUT, there is a strange thing.
HITL only works when it is the first time HITL after flash new firmware. Strange, right?
It's like this
| flash Firmware
| HITL (working!)
`MSG: RECOVERING FROM FMU IN-AIR RESTART
MSG:Failed to recover from in-air restart (1), abort
MSG:RECOVERING FROM FMU IN-AIR RESTART
MSG:ARMED by arm/disarm component command
MSG:Failsafe enabled:no RC
Init MAVLink`
And after arm, it works fine!
[image]
[image]

| close HITL, and reboot vehicle
| HITL (not working)
[image]

| reboot again
| HITL (never working again)

Based upon: v2, I use make xxx force-upload to upload it. (my hardware has 2mb flash)

-----------------------Line----------------

I use
`make px4_fmu-v3_default upload` (I can use v3, because i followed [update bootloader](https://docs.px4.io/en/advanced_config/bootloader_update.html). Evan though QGround still shows I'm using fmu_v2, but when I 'make px4_fmu-v2_default upload' It tells me should use "px4_fmu-v3_default")

Anyway, HITL works anytime! So it solved for me.
Thanks for your help.
If interesting, strange problem above the --line-- can be looked into.

julianoes, 2019-01-17:
> | HITL (not working)

When HITL is not working, does also unplugging it and re-plugging it not work?

closed 2019-07-01 by julianoes