PX4/PX4-Autopilot #11213: Tiltrotor SITL interrmittant failure

dagar, 2019-01-14:
Originally reported in https://github.com/PX4/Firmware/issues/11144

New failures -
 - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/pr-2swhoop-params/3/pipeline

> Different issue, tiltrotor arming failing due to accels. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-11136/4/pipeline/81
> 
> [image]
> 
> What's going on with the vibration in the tiltrotor model?
> 
> From the failure log - https://logs.px4.io/plot_app?log=33db160e-75a1-413c-9401-9bda6daa063b
> [image]
> 
> From a successful log - https://logs.px4.io/plot_app?log=a20570ad-bd94-446d-b910-f56e441610b5
> [image]

@RomanBapst any ideas?

dagar, 2019-01-21:
More failures.
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/pr-update_src%2Flib%2FDriverFramework/5/pipeline/80

lamping7, 2019-03-15:
Good to close @dagar?

dagar, 2019-03-16:
Actually no, I've seen it a couple times in the last week or so. I'll record any new occurrences here.

dagar, 2019-03-16:
Here's a fresh one. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-11668/5/pipeline

[image]

EliaTarasov, 2019-03-28:
@dagar i think it's just the P-component for the tilting actuators is too large. Oscillations due to high P-values causes high-frequency jumping on the ground and sometimes accelerometer range becomes too large. I set P equals 10 and this issue disappeared.

julianoes, 2019-04-26:
Looks like #11912.

PR #11917 (2019-04-28): Fix standard_vtol vibration on ground
This addresses the issue where the `standard_vtol` (delta quad) vibrates on the ground.

Also, this moves the airspeed noise from `simulator_mavlink.cpp` to the Gazebo side.

Fixes #11912, fixes #11213.
closed 2019-04-28 by dagar