PX4/PX4-Autopilot #11217: Multi-Vehicle Simulation Error

hxdnshx, 2019-01-15:
**Describe the bug**
I executed "roslaunch px4 multi_uav_mavros_sitl.launch", and got following error:

> while processing /home/user/Firmware/launch/single_vehicle_spawn.launch:
Invalid <arg> tag: xacro
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
ROS path [2]=/home/user/Firmware
ROS path [3]=/home/user/Firmware/Tools/sitl_gazebo
ROS path [4]=/home/user/Firmware
ROS path [5]=/home/user/Firmware/Tools/sitl_gazebo.
> 
> Arg xml is \<arg default="$(find xacro)/xacro $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port) --inorder" name="cmd"/>
The traceback for the exception was written to the log file

**To Reproduce**
Steps to reproduce the behavior:
1. Luanch a px4 docker container
>docker run -it --privileged -p 14562:14556/udp --name=mycontainer px4io/px4-dev-ros:lat
est bash
2. Clone px4 firmware
>git clone --recursive https://github.com/PX4/Firmware.git
3. Build,Source and launch(From [documention](https://dev.px4.io/v1.8.2/en/simulation/multi-vehicle-simulation.html))
>cd Firmware/
>make posix_sitl_default
>make posix_sitl_default sitl_gazebo
>source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
>export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/sitl_gazebo
>roslaunch px4 multi_uav_mavros_sitl.launch

lamping7, 2019-01-22:
Whats the output of `rospack find xacro` in your container?

hxdnshx, 2019-01-23:
Thanks for your consideration. Everything works well after i installed package `ros-kinetic-xacro`.
So i created a [pull-request](https://github.com/PX4/containers/pull/158) to the project PX4/container.

lamping7, 2019-01-23:
Perfect. That was my guess.

closed 2019-01-23 by lamping7