PX4/PX4-Autopilot #11221: SBUS out and I2C sensors not working

multimensch, 2019-01-15:
Hi,
I have a Pixhawx 4 with 1.8.2 firmware/QGroundControl and this is the first time I am using it. My Hexacopter is flying well with the Pixhawk 4 and I wanted to add some additional features, but without success.
My Problems are:
- I don´t get a signal on the sbus out port
- I2C Sensors are not working/recognized (Lidar-Lite (LL40LS), Maxbotix Sonar (mb12xx))
I set up the corrosponding parameters but nothing works. I don´t find any other settings.
Can someone support me with some information how to set up these things?

DanielePettenuzzo, 2019-01-19:
For sbus out did you set PWM_SBUS_MODE to 1. Moreover on this output you will get the same output that you would get on the IO PWM port. Therefore IO must be enabled (SYS_USE_IO = 1) and you should have a mixer loaded on this output. 

Regarding the I2C sensors the automatic scanning of ports I2CA and I2CB is supported on master but not on the latest release. This will be included in the next stable release. You can try either `master` or `v1.9.0-alpha`

DanielePettenuzzo, 2019-01-24:
What distance sensor on which i2c port? I will try to reproduce the issue. Can you also send me the output of the commands on the mavlink console:
pwm info 
px4io status

multimensch, 2019-01-28:
On Saturday I accidentaly flashed 1.9.0 beta and everything is working as expected now. Lidar-Lite, Maxbotix Sonar and PX4Flow. I am so happy now, thanks.
SBUS OUT is still not working and the PWM_SBUS_MODE parameter disappeared.

DanielePettenuzzo, 2019-01-28:
Happy your sensors are working. Regarding the PWM_SBUS_MODE parameter it might be that the px4io is not running. Can you check the SYS_USE_IO parameter if it’s enabled? Then it would help to see the outputs of the following mavlink console commands after you enable the IO:

pwm info 
px4io status

The mavlink console can be access via qgc

multimensch, 2019-01-29:
The PWM_SBUS_MODE parameter appeared after setting up the vehicle type.
PWM_SBUS_MODE = 1 and SYS_USE_IO is enabled but still no SBUS output.

[image]
[image]

DanielePettenuzzo, 2019-01-29:
It looks like your SBUS is running on the SBUS_OUT port on the pixhawk. Are probing it with a logic analyzer?

DanielePettenuzzo, 2019-01-31:
I will try to reproduce this. What airframe did you choose from QGC?

multimensch, 2019-02-12:
Sorry I was on vacation...
I use hexarotor x.

julianoes, 2019-07-10:
@DanielePettenuzzo could you follow up?

@multimensch have you tried v1.9 in the meantime?

FlavioTonelli, 2019-07-26:
I can confirm sbus out is working as expected on Pixhawk Cube and PX4 1.9.2.
I can correctly read the same pwm values of the eight main output.

closed 2019-07-29 by julianoes

dennisonlau, 2019-11-15:
@FlavioTonelli Hi I am trying to read the pwm values from Pixhawk through the sbus. Did you use any library to decode it?

FlavioTonelli, 2019-11-16:
Hello @dennisonlau. I used two Pixhawk, one generating sbus signal and one reading the signal as a classic RC input one.
Of course is not the best way, but it was on my side the less expensive on a time base.
I can't help you much more, but there are plenty of ways to read such a signal. A classic board with a micro can easily do that task.

Last but not least, this platform is intended as a bug tracker. I suggest to use px4 forum or community channel to receive further assistance.

dennisonlau, 2019-11-16:
Hi @FlavioTonelli  Thanks for the quick reply. Do you know which repository consist of the code for implementing the sbus out protocol? I am currently reading the port with Saleae logic but the bit pattern does not match the sbus protocol I originally knew. 

Thanks

FlavioTonelli, 2019-11-16:
I'm not sure it can help you but you can try to have a look at src/lib/rc/sbus files