PX4/PX4-Autopilot #11222: Tricopter spinning uncontrollably

tecdee, 2019-01-15:
Hi guys,

I have a tricopter running the Pixahwk 1 autopilot. I have setup everything correctly. I am using the Tricopter Y+ airframe where all the rotors spin counterclockwise. The tricopter arms correctly, but as soon as I try to fly it, it spins/rotates uncontrollably in the clockwise direction. 

What I have attempted:
- I have reversed the servo channel via the RC4_REV parameter in QGC. That didn't work.
- I have also reversed the Rudder channel on my Radio Tx, also to no avail.

These are the steps which solved this issue for some other users found via my online search, but its not working for me. 

Please help.

Thanks.

tecdee, 2019-01-16:
So here's some logs from manual tests I've done. I simply held the copter in hand and tried to spin it in different directions. These logs are with different PID gains. 

https://review.px4.io/plot_app?log=4dfa5c20-b1a1-44e7-ab9e-2dd0390480c6

https://review.px4.io/plot_app?log=5b17d57a-73ea-42f5-9028-bf1ccd12769b

Thanks.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
> vibrations are massive.

I suspect that's a hardware or tuning issue.

deehat, 2021-02-01:
@julianoes thank you. Yes the vibration is a hardware issue.