PX4/PX4-Autopilot #11225: Add support for STATUSTEXT_LONG

hamishwillee, 2019-01-16:
[STATUSTEXT_LONG](https://mavlink.io/en/messages/common.html#STATUSTEXT_LONG) was added to the MAVLink spec. It allows longer than 40 char status messages. 

My assumption was that PX4 would use this in preference to [STATUSTEXT](https://mavlink.io/en/messages/common.html#STATUSTEXT) except when talking to a MAVLink 1 system, and that QGC would support reading both messages.

hamishwillee, 2019-01-17:
FYI, support for this goes into QGC in https://github.com/mavlink/qgroundcontrol/pull/7152

hamishwillee, 2019-06-27:
FYI, not sure whether or not this is needed because the event's interface should be used for many of the use cases that justified this message's existence.

hamishwillee, 2019-09-25:
On the other hand ...

hamishwillee, 2020-01-05:
I think next time round will allow this to lapse. We're now coming up to the point where events interface should exist.

julianoes, 2020-02-14:
STATUSTEXT_LONG  is dead, next come chunks:
https://github.com/mavlink/mavlink/pull/1295

hamishwillee, 2020-02-17:
Thanks. Closing this, and opening a new one: https://github.com/PX4/Firmware/issues/14169

closed 2020-02-17 by hamishwillee