PX4/PX4-Autopilot #11227: receiving incomplete data from PX4Flow to PX4 fmu-v5

60ericleal60, 2019-01-16:
I forked the PX4 repository on Jan 10 2019 to add some air frames and uploaded that to my Pixhawk 4. I cloned the PX4Flow repository and uploaded it to the optical flow sensor today the 16th. When the PX4Flow board is connected to my computer over USB to QGC all looks good but when it is going through the Pixhawk through the I2C bus I only get some of the data with mostly 0's

[image]

I then hooked the PX4Flow board to a Nucleo f767zi and read the outputs with a simple read and display of the 26 byte register 0x16 and all of the data came through correctly. Any ideas of whats going on and where I should look?

mhkabir, 2019-01-17:
Can you try unplugging and replugging the flow module without rebooting the pixhawk and see if the data starts coming through?

60ericleal60, 2019-01-17:
Ok so when I do that I will get 3 or 4 readings in a row and then it goes back to zeros.

60ericleal60, 2019-01-22:
I uploaded the Ardupilot firmware to the PX4Flow board from QGroundControl to see if the readings to the NUCLEO board were any different from the 0x16 register and it was not. However when I connected the PX4Flow board with the Ardupilot firware to the Pixhawk with the PX4 stack, all of the data comes through to the MAVlink inspector. I also tried putting the stable release of PX4 1.8.2 on the Pixhawk and optical flow does not show up in the Mavlink inspector at all.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
@60ericleal60 have you found a solution in the meantime?