PX4/PX4-Autopilot #11235: the wind static barometric pressure correction

int32cn, 2019-01-18:
At EFK2_main.cpp run function, the wind static barometric pressure correction is calculated as following:

const float pstatic_err = 0.5f * airdata.rho *
    (K_pstatic_coef_x * x_v2) + (_K_pstatic_coef_y.get() * y_v2) + (_K_pstatic_coef_z.get() * z_v2);

And according the air static barometric pressure formula，P = 0.5 * ro * V ^2;
So I 'm wondering here maybe missing a brace, the code above should be as the following:
const float pstatic_err = 0.5f * airdata.rho *
   ( (K_pstatic_coef_x * x_v2) + (_K_pstatic_coef_y.get() * y_v2) + (_K_pstatic_coef_z.get() * z_v2));

CarlOlsson, 2019-06-25:
This is still an issue.
Pinging @RomanBapst just so it does not get forgotten

RomanBapst, 2019-06-26:
@CarlOlsson thanks for mentioning this, I will fix it today.

RomanBapst, 2019-06-26:
@int32cn Thanks for raising this!

PR #12332 (2019-06-27): ekf2: fixed calculation of static pressure error
Fixes #11235
closed 2019-06-27 by bresch