PX4/PX4-Autopilot #11240: Ekf2 doesn't use data from sensors immediatly

Slimercorp, 2019-01-19:
**Describe the bug**
EKF2 timestamps reports that ekf2 doesn't use mag, baro, gps and other sensors immediately (when topic was updated). EKF2 uses it after 1-2 self iterations. Why?
For example:
0 iteration: IMU was updated (timestamp 4000), mag wasn't updated. EKF2 use IMU data only.
1 iteration: IMU was updated (timestamp 8000), mag was updated (timestamp 6500). EKF2 use IMU data only.
2 iteration: IMU was updated (timestamp 12000), mag not updated (timestamp 6500). EKF2 use IMU and mag data.

It may see on relative timestamp. If relative timestmap > 40, then it means ekf2 doesn't use data immediately.

The problem meets in all logs, that i can see here http://logs.px4.io/.

**To Reproduce**
Use, for example, firmware v1.8.2. Look on mag relative timestamp.

**Expected behavior**
Perfectly all relative timestamps must be lower 40 (4000 microseconds, it's time of update IMU and ekf2). Other words, new data from mag, baro, gps sensors use immedeatly

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=1a0d7681-2c9d-462c-877f-f4338718de56
[image]

Where is explanation of this problem?

g0ro, 2019-01-29:
See https://ardntech.atlassian.net/wiki/spaces/FLYP/pages/19529770/EKF2

priseborough, 2019-02-04:
EKF state prediction and fusion cycle is running on a delayed time horizon as explained here: https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html

closed 2019-02-04 by priseborough