PX4/PX4-Autopilot #11257: FW SITL intermittent failure

dagar, 2019-01-21:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master_nuttx_7.28%2B/3/pipeline

Exiting early?
[image]
https://logs.px4.io/plot_app?log=214c70d1-d6c9-4cf7-a655-5faab83328e1

dagar, 2019-01-21:
@TSC21 @julianoes any ideas? This one is popping up pretty often. Let's at least collect examples in this issue as they appear.

julianoes, 2019-01-21:
Ok, I'll try to reproduce this locally tomorrow.

julianoes, 2019-01-22:
Which test exactly is failing? I can't access the Jenkins log anymore.

dagar, 2019-01-22:
The first one (FW).

[image]

dagar, 2019-01-22:
> Not exiting early on the one I looked at (3 in your list). The test is terminating after the 5 minute timeout. This is confusing because the output in Jenkins isn't necessarily in order. Here's the error from the test that kills the simulation:
> 
> > ERROR: max time [300.0s] allotted for test [FW_mission_1] of type [px4/mission_test.py]
> 
> Something is crashing before the test starts:
> 
> > Unhandled exception in thread started by
> > sys.excepthook is missing
> > lost sys.stderr
> 

Yes this is actually what I meant by exiting early (gazebo side?).

julianoes, 2019-01-23:
I ran this about 20 times in my Virtual Machine and I couldn't reproduce this failure. I'll keep trying.

mcsauder, 2019-01-24:
@julianoes , it's possible that this issue might be user behavior related.  I just realized that I have consistently been rebasing multiple PRs and force-pushing to remote within minutes of each other.  I assume this is kicking the CI tools into queuing multiple runs back-to-back.  Maybe this difference has an impact.

lamping7, 2019-01-24:
We need to determine what changed in sitl (likely outside of the firmware) that introduced this.

lamping7, 2019-01-24:
@julianoes I ran this multiple times in CI with gazebo `debug` flag set. I'm seeing pages of:
>[Err] [gazebo_mavlink_interface.cpp:1018] poll timeout
Any ideas how to debug further?

lamping7, 2019-01-24:
related: https://github.com/PX4/Firmware/issues/11144

mcsauder, 2019-01-25:
thanks @lamping7 , and @julianoes ... I've been seeing this for at least the past 6 months. I just haven't found any ways to improve things so I haven't spoken up about it.

lamping7, 2019-01-25:
Crazy idea... The `plane` model is the only one with a camera (among the others we test in Jenkins). Is the lack of display crashing Gazebo?

julianoes, 2019-01-28:
@lamping7 good point. I have sometimes had spurious Gazebo crashes with some message about GLX graphics whatnot. So that could be the same thing.

julianoes, 2019-01-29:
I compared a [successful test](http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-11310/2/pipeline/75) with the [failed one](http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master_nuttx_7.28%2B/14/pipeline/75) and found that in the one passing, all (MAVROS or ROS?) plugins seem to be loaded before the PX4 output (including "Exiting...") appears whereas in the failing one, some of the plugins are loaded later.

However, I don't know if these logs actually appear in order or if that has nothing to do with the actual timeline.

lamping7, 2019-01-29:
They do not appear in order. Jenkins is magical like that.

dagar, 2019-02-05:
I think the main failure this issue was opened to address has been resolved. Continue the "poll timeout" failure in https://github.com/PX4/Firmware/issues/11380.

closed 2019-02-05 by dagar

julianoes, 2019-02-06:
What exactly fixed this? Removing the camera?

julianoes, 2019-02-06:
> Mostly, yes.

What does that mean?