PX4/PX4-Autopilot #11275: Level horizon calibration transients

RomanBapst, 2019-01-22:
**Describe the bug**
It seems like the board rotation parameters are set to zero prior to performing the attitude calibration. In some cases this can cause attitude transients which impact the average measured attitude.

I'll follow up with a plot if not already with a fix.

RomanBapst, 2019-01-22:
@sanderux FYI

julianoes, 2019-07-15:
@RomanBapst is this still an issue?

julianoes, 2019-10-14:
@RomanBapst it would be good if you can follow up on these things, thanks.

RomanBapst, 2019-10-18:
@julianoes This problem was reported by @sanderux and I created this issue so that we don't forget about it. I have just tried to do a level calibration with a 30 degrees offset on both roll and pitch and the level horizon calibration worked just fine. The algorithm was waiting long enough for the transient of the attitude solution to decay.
Given that this is the only issue I've ever seen about this and I cannot reproduce the problem I'm going to close this.
@sanderux If you have a log of a failure case please share it here and we can have a look at it.

closed 2019-10-18 by RomanBapst

RomanBapst, 2019-12-03:
@julianoes @sanderux  This should now be fixed for good since @bkueng did changes to the calibration routine such that the calibration is performed relative to the current board offset.
https://github.com/PX4/Firmware/pull/13666#pullrequestreview-326119932