PX4/PX4-Autopilot #11280: CI ROS tests too hard to reproduce

julianoes, 2019-01-23:
TL;DR: We need more cohesive docs and scripts to run the CI tests.

To investigate #11257 I set out to reproduce the CI tests on my local machine. Running it in docker was out of the question because it makes catching hardfaults even harder (I tried that unsuccessfully last time).

Therefore, fresh VirtualBox with Ubuntu 16.04, and then I try to follow the instructions as well as possible.

```
wget https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_nuttx.sh
chmod +x ubuntu_sim_nuttx.sh
./ubuntu_sim_nuttx.sh
```

So far so good, then I needed the ROS dependencies:

```
wget https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_gazebo.sh
chmod +x ubuntu_sim_ros_gazebo.sh
./ubuntu_sim_ros_gazebo.sh
```

Here I ran into troubles because of Gazebo 9 being installed by `ubuntu_sim_nuttx.sh` but `ubuntu_sim_ros_gazebo.sh` needs Gazebo 7 (see separate [issue](https://github.com/PX4/Devguide/issues/722). I had to remove Gazebo 9, and then run it again.

```
sudo apt-get remove gazebo7 libgazebo7-dev
./ubuntu_sim_ros_gazebo.sh
```
Now, I should be able to run the tests, confusingly the clone is already here, that's magic:

```
cd src/Firmware
git submodule update --init --recursive
test/rostest_px4_run.sh mavros_posix_test_mission.test mission:=FW_mission_1 vehicle:=plane
```

Ok, looks like MAVROS is missing somehow:
```
GAZEBO_PLUGIN_PATH :/home/julianoes/src/Firmware/test/../build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/julianoes/src/Firmware/test/../Tools/sitl_gazebo/models
LD_LIBRARY_PATH /opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/julianoes/src/Firmware/test/../build/px4_sitl_default/build_gazebo
... logging to /home/julianoes/.ros/ros_logs/rostest-ubuntu1604-vbox-6442.log
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rostest", line 36, in <module>
    rostestmain()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/__init__.py", line 274, in rostestmain
    _main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_main.py", line 159, in rostestmain
    testCase = rostest.runner.createUnitTest(pkg, test_file, options.reuse_master, options.clear, options.results_base_dir)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 229, in createUnitTest
    config = roslaunch.parent.load_config_default([test_file], None)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
[... 193 lines cut ...]
[ INFO] [1548227426.795887203]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[INFO] [1548227427.360610, 0.004000]: Calling service /gazebo/spawn_sdf_model
[ INFO] [1548227427.381234468, 0.024000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1548227427.614622, 0.184000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1548227427.641899045, 0.184000000]: Physics dynamic reconfigure ready.
Traceback (most recent call last):
  File "/home/julianoes/src/Firmware/integrationtests/python_src/px4_it/mavros/mission_test.py", line 52, in <module>
    from mavros import mavlink
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/mavros/mavlink.py", line 12, in <module>
    from pymavlink import mavutil
ImportError: No module named pymavlink
INFO  [simulator] Simulator connected on TCP port 4560.
10.10
ERROR [px4] Startup script returned with return value: 2
pure virtual method called
terminate called without an active exception
```

Ok, so pymavlink is also missing.

```
pip install --user pymavlink
```

And we're flying, finally!

lamping7, 2019-01-23:
I know you know this is related, but want to add it here: https://github.com/PX4/Firmware/issues/11024
Also: https://github.com/PX4/containers/pull/107

Normally I use the containers that CI uses to rule out setup differences.

I also noticed that the script you mentioned installs mavros from source, which is a topic that could be discussed.

If we get containers and Jenkins to use these, we can actually maintain these and know when they break.

julianoes, 2019-06-04:
I'm fighting with this again, this time using docker where `rostest` seems to be missing.

julianoes, 2024-03-04:
Old, closing.

closed 2024-03-04 by julianoes