PX4/PX4-Autopilot #11281: Some topics are being published only after 50 s in replay

CarlOlsson, 2019-01-23:
**Describe the bug**
When replaying most topics, e.g. ekf2_innovaitons, are only published after ~40 seconds. In contrast estimator_status is published from the beginning
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. replay any logfile

**Expected behavior**
The logged data match while replaying should match the data in the original log

**Additional context**
I found that if the ekf2_innovations topic is advertised at the start of the replay (at the same place as estimator_status) and not only when it is being published for the first time it seems to work as expected

[image]

Maybe @bkueng knows if this is expected

bkueng, 2019-01-23:
Yes, the logger only checks for new topics every second after startup: https://github.com/PX4/Firmware/blob/master/src/modules/logger/logger.h#L52.
Probably easiest to advertise the topic directly on startup, since the topic is published eventually anyway.

CarlOlsson, 2019-01-23:
but if it checks every second, why does it take 40 seconds until it is logged?

bkueng, 2019-01-23:
The replay runs much faster, and the logger uses the real system-time for the check.
Now that we have time abstraction, we could actually use it for the replay as well (@julianoes), then this would not happen either.

CarlOlsson, 2019-01-23:
ah okey now I get it, thanks for the explanation (Y)

CarlOlsson, 2020-01-15:
I just tested v1.10 and this issue is still present, is anyone using ekf replay without changing ekf2_main to advertise the topics at bootup?

bkueng, 2020-12-22:
Replay is now built with lockstep (https://github.com/PX4/PX4-Autopilot/pull/15137), so this should be resolved. Please reopen if not.

closed 2020-12-22 by bkueng