PX4/PX4-Autopilot #11296: Oversampling rate for MPL3115A2 pressure sensor

LeoMustafaNXP, 2019-01-25:
**Describe the bug**
On my test flights I see that the drone is not able to keep height and jumps +-2m up and down. 
Parameters?

In the log files I saw that the Barometer Altitude is very noisy. In the driver for the MPL3115A2 the default Oversampling rate is set to 0. This means that the minimum time between data samples is 6ms. I have done some test flights with different OSRs and it seems to be better to use as default OSR 2. This means that the minimum time between data samples is 18ms. In outdoor test flights the drone was able to hold the height better and jumps only +-0.5m up and down. 

Please change the default OSR from 0 to 2. 
(File: /src/drivers/barometer/mpl3115a2.cpp line 100)
#define MPL3115A2_OSR 2

**To Reproduce**
FMU: NXP FMU K66 
Drone: LJI500 frame
GPS: Pixhawk 4
Telemetry: Holybro 433MHz radio

**Log Files and Screenshots**

OSR 1: https://review.px4.io/plot_app?log=c90e1169-a56f-4694-8372-628e4567865c
[image]

OSR 2: https://review.px4.io/plot_app?log=aaefb21d-73c8-4427-9ea6-aa92bc8a56c9
[image]

OSR 3: https://review.px4.io/plot_app?log=41b6562d-f1de-4e50-9062-81d98e9012b5
[image]

OSR 4: https://review.px4.io/plot_app?log=88647968-63ee-48d0-ab29-e055ddd84098
[image]

dagar, 2019-01-25:
Seems like a clear improvement. Any input on this @priseborough or @davids5?

davids5, 2019-01-29:
Thank you @LeoMustafaNXP  and @priseborough PR is in. #11332

dagar, 2019-01-29:
Fixed in https://github.com/PX4/Firmware/pull/11332

closed 2019-01-29 by dagar