PX4/PX4-Autopilot #11300: local_position_estimator and position_estimator_inav basic support or deprecation

dagar, 2019-01-25:
Currently EKF2 (ecl/EKF) is the default estimator for all platforms, but we still have the local_position_estimator (LPE), and position_estimator_inav in tree. Likely both still have small userbases. 

LPE can be enabled with the `SYS_MC_EST_GROUP` parameter.

I feel strongly that we either need to add a minimal support baseline for these estimators (simple testing) or start deprecating them. We're not helping users or the perception of PX4 in general by dragging things along that are unsupported, and quite possibly unsafe to use.

### Options

1. Minimal baseline for upstream inclusion
   1. simple SITL tests
   2. occasional flight test
   3. volunteer willing to commit to reviewing all LPE and INAV labelled issues
   4. (later) new c++ interface to enforce estimator publishing requirements.
      - example: many parts of the system use the `sensor_bias` (https://github.com/PX4/Firmware/blob/master/msg/sensor_bias.msg) message, but neither LPE or INAV publish it.

2. Begin deprecation
   1. move to examples
   2. remove or modify SYS_MC_EST_GROUP parameter to prevent direct user usage
   3. delete entirely after a major release or two in examples

dagar, 2019-01-25:
I'd like to implement one of the options before the next stable release of PX4 (v1.9.0).

TSC21, 2019-01-25:
On it. I will start with LPE and set a MC mission box test.

LorenzMeier, 2019-01-27:
I think its time to deprecate - the level of QA on these is really poor compared to the rest of the system and its becoming a reputation risk. We can't drag them along if there are not developers using them and really depending on them. I would remove INAV for 1.9.

hamishwillee, 2019-01-29:
FYI only, this has a docs implication in the devguide.

dagar, 2019-10-13:
We've dropped position_estimator_inav.

closed 2019-10-13 by dagar