PX4/PX4-Autopilot #11317: Tangent Loiter exit not working correctly for Fixed Wing loiters

Antiheavy, 2019-01-28:
The tangent loiter exit criteria is not working for Fixed Wing loiters (as tested in SITL).

Using the exact same mission plan, v1.8.0 and Master have different behavior.  v1.8.0 correctly waits until the vehicle heading matches the track to the next waypoint.  Master incorrectly breaks out of the loiter immediately and turns to the next waypoint.

[image]

Antiheavy, 2019-02-04:
Tangent exit also not working for time based loiters.  see SITL screenshot:
[image]

ryanjAA, 2019-02-13:
Flew this today with a bixler on master from today. See screenshot. Trying to get the logs - seem not to be playing nice with new SD cards. Will retrieve telem log if need be. Bear in mind there was about 10 knot winds sustained but still shouldn't make exits anything but less stable, not in the wrong location. Specifically the last exit was a fixed wing landing pattern, not even close to the right direction. Interestingly, the fixed wing loiter to land on the previous flight was fine, only difference was we had not loiter at all in the previous flight.

[image]

ryanjAA, 2019-02-13:
[Bixler day 2.plan.zip](https://github.com/PX4/Firmware/files/2858215/Bixler.day.2.plan.zip)

dannyfpv, 2019-02-13:
today test master on phantom wing pixhawk 1 from waypoint loiter to rtl home point
https://review.px4.io/plot_app?log=1c6b6a1a-69cf-4c5e-8285-624239a4ceaa

Antiheavy, 2019-02-13:
Thanks @dannyfpv .  Question: did you use a standard Fixed Wing Landing pattern from QGC in that flight?  If not, did your loiter waypoint have tangent exit criteria set?  If not either, I'm not sure if you actually tested the "bug".
[image]

ryanjAA, 2019-02-24:
So i'm a bit torn on this. Heres why. This log [https://review.px4.io/plot_app?log=e6a48d76-9a8b-448b-8592-8df8525928cc](url) shows leaving not tangent. And we did this a bunch of times so there are plenty more to show. It leaves quasi tangent, in that it's tangent to the circle but in the completely wrong area. See image 1. Then this kept happening and we adjusting the altitudes, etc and it didnt matter but getting a strange exit like this was not good. It landed fine because it would turn back left (like in the image) but still, not ideal. So then we moved the fixed wing landing back quite a bit. See image 2 and log 2 below. This did make it exit tangent and land but both scenarios should have (i think at least) exited the same way). The more we flew this, im in agreement, something isn't right. It cant work one way sometimes and another way others and honestly, what's the point of the loiter to alt exit tangent on a landing if it's not predictable and leaving out the loiter in the wrong way. 

My vote it something is wrong after spending a bit of time looking at these. We have quite a few logs from Friday so happy to share if more are wanted but this kept happening and even when we were bixler testing early in week, the loiter landing did the same thing.  Log 2 is here [https://review.px4.io/plot_app?log=be07358a-a4ac-4ba1-8e2b-299410c35c51](url)

[image]
[image]

ryanjAA, 2019-02-24:
Looking back at the top. The image from SITL is literally exactly what is happening on my previous post (image 1) and my post from 10 days ago.

This is broken as far as im concerned and why moving the loiter to landing back gets it to do it correctly is also strange.

Antiheavy, 2019-02-24:
Thanks for the real flight tests @ryanjAA .  I don't see any difference in SITL with long or short landing approaches.  Sometimes you just get lucky and it looks like it is exiting on the tangent at the correct heading, but really it seems to be exiting the loiter as soon as altitude is reached and ignoring the heading requirement.

I've finally SITL tested with v1.8.2 stable and it tests fine.  So this was broken after 1.8.2

LorenzMeier, 2019-02-24:
@dagar It shouldn't be too hard to fix the loiter exit to wait for the tangent - could you look into that or should I coordinate with @bresch?

TSC21, 2019-02-24:
@LorenzMeier I will be looking into this as well.

dagar, 2019-02-25:
Likely something broken in the somewhat complex LOITER_TO_ALT interaction (navigator mission block).

PR #11576 (2019-03-04): navigator fix FW loiter tangent exit
The yaw acceptance feedback from the position controller (introduced for MC collision avoidance) is always NAN for fixedwing.

 - fixes #11317

Needs testing.
commit 468fb534 (2019-03-04): navigator fix FW loiter to alt tangent exit (#11576)

 - fixes #11317
closed 2019-03-04 by dagar