PX4/PX4-Autopilot #11320: Organize codes of px4_raspberrypi_* in developer specific folder?

SalimTerryLi, 2019-01-28:
Due to the condition that px4_raspberrypi_* and emlid_navio were splitted, it' s possible to maintain the pure branch of raspi. But developers who work on raspi without navio HATs mostly work separately with their own configuration of hardware.

Now the board specific code are moved into boards/ . I' m asking to create something like: 
boards/px4/raspberrypi/[developer's name]/src/    // for developer' s personalized code
boards/px4/raspberrypi/src/    // for reusable codes

Or, any maintainer focused on this branch?

dagar, 2019-01-28:
What's an example of generic raspberry pi code with no appropriate home?

SalimTerryLi, 2019-01-28:
A direct example is src/drivers/rpi_rc_in/ . It was merged incompletely from someone' s project, missing some extern codes. Besides, adc & actual control & rc input & some other sensors will perform their function in unpredictable forms too. Though these extra sensors have not been employed, it woull be a possibility. Raspi platform has too much freedom to expand its function, but its developers rarely work on the same branch.
I hope that codes can be reused by other platform but at first they should have a place to exist.

SalimTerryLi, 2019-01-28:
> 
> 
> What's an example of generic raspberry pi code with no appropriate home?

A direct example is src/drivers/rpi_rc_in/ . It was merged incompletely from someone' s project, missing some extern codes. Besides, adc & actual control & rc input & some other sensors will perform their function in unpredictable forms too. Though these extra sensors have not been employed, it woull be a possibility. Raspi platform has too much freedom to expand its function, but its developers rarely work on the same branch.
I hope that codes can be reused by other platform but at first they should have a place to exist.

dagar, 2019-01-29:
Wouldn't that be an example of something that's effectively only for the plain raspberry pi (currently px4/raspberrypi)? Something with a navio2 already has PPM in through sysfs.

SalimTerryLi, 2019-01-30:
> 
> 
> Wouldn't that be an example of something that's effectively only for the plain raspberry pi (currently px4/raspberrypi)? Something with a navio2 already has PPM in through sysfs.

Yes, that is what I' m talking about. Navio2 has its own fixed hardware configuration, which forms its development. But the plain raspberry pi does not. Developers can provide many kinds of solutions they prefered.

closed 2019-05-24 by SalimTerryLi