PX4/PX4-Autopilot #11333: Pixhawk 4 Mini QGC Detected 0 Channels Issue with every RX

ludoplt, 2019-01-29:
Hello, 

I start configuring a Pixhawk 4 Mini from Holybro in QGC. Firmware version is 1.8.2.
Everything goes well for sensors calibration but I wasn't able to get RC signal. 

I check my cable, connexion is OK, in the right place in the right orientation !

I test with R7008SB / another R7008SB and R2008SB RXs with T14SG TX.
I also test with X8R RXs with Taranis QX7 TX, same error ! 

What am I missing ? Is my Pixhawk unit broken ? 

Thanks to who can help me !
Ludo

ludoplt, 2019-01-29:
For information, it works well with master release 1.9.0beta but not with stable release...

bys1123, 2019-02-11:
Which port you connecting on Pixhawk 4 Mini

ludoplt, 2019-02-18:
The RC IN port, not the PPM.

closed 2019-07-08 by stale[bot]